espressif/esp-idf #15796: Pleasew clarify USB-HID hardware support

jumpjack, 2025-04-16:
### Is your feature request related to a problem?

"Note: In case your board doesn't have micro-USB connector connected to USB-OTG peripheral, you may have to DIY a cable and connect D+ and D- to the pins listed below." ( https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/README.md#common-pin-assignments )

What does this mean? How many USB cables should I use to connect ESP32 to PC to emulate keyboard? Or, how do I know if an ESP32 dev board variant does support USB HID? I need a plug-and-play solution, no soldering.

tore-espressif, 2025-04-17:
hello @jumpjack 
For emulating USB HID device, you will need ESP32-S2/S3 development board that has 2 USB connectors - one is for programming, the second is for your USB HID function. For example this board https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitc-1/user_guide.html

Only some older S2 devkits do not have the second USB connector

jumpjack, 2025-04-18:
So no -mini variant is suitable, which is what I need because I need a very tiny variant.

igrr, 2025-04-18:
Here is a board with an ESP32-S3-MINI-1 module: https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32s3/esp32-s3-devkitm-1/user_guide.html#esp32-s3-devkitm-1

jumpjack, 2025-04-19:
it's huge, I need something 3x3cm at most.

I just received this one, purchased quite "randomly" and "hopefully"... and it works!

https://www.amazon.it/dp/B0BGPD6CV5

It has just one USB port, but it reconfigures depeding on how you press buttons: 0+Reset puts it into USB-OTG, Reset+0 puts it in standard (serial?) mode.

chegewara, 2025-04-23:
> it's huge, I need something 3x3cm at most.
>
> I just received this one, purchased quite "randomly" and "hopefully"... and it works!
>
> https://www.amazon.it/dp/B0BGPD6CV5
>
> It has just one USB port, but [...]

This module is not using USB to UART like CP21xx to communicate with esp32, instead it is using internal hardware USB (pins D+/D-).
This has cons, like when you are working with tinyUSB or USB host code then you cant output logs to that USB port, but you need external USB to UART device.
You can find more devkits/boards like this.

tore-espressif, 2025-04-29:
@jumpjack I'll close this issue as the answer was provided. Feel free to reopen if you have more questions

closed 2025-04-29 by tore-espressif