espressif/esp-idf #15798: esp_video simple_video_server build error

truck-iguana, 2025-04-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

[I am referencing this espressif component](https://components.espressif.com/components/espressif/esp_video/versions/0.9.0~1)

[Specifically, I am using this example](https://components.espressif.com/components/espressif/esp_video/versions/0.9.0~1/examples/simple_video_server)

I expect to be able to build the project according to the regular process.

```
idf.py create-project-from-example "espressif/esp_video:simple_video_server"
cd ./simple_video_server
idf.py build flash monitor
```

This fails.

### What is the actual behavior?

[I have referenced the idf-component-manager](https://docs.espressif.com/projects/idf-component-manager/en/latest/)
`compote version` returns `2.1.2`
I've run the following commands

```
idf.py create-project-from-example "espressif/esp_video=0.9.0~1:simple_video_server"
```
As denoted in the example linked above.
This returns the error
```
ERROR: Invalid version specification: "=0.9.0~". Please use format like ">=1" or "*"
```
I got around this with
```
idf.py create-project-from-example "espressif/esp_video>=0.9.0:simple_video_server"
```
The following build then fails.
```
idf.py set-target esp32 # reports out some errors
idf.py build # reports the same CMake build errors
```

The NOTICE's and ERROR are shown below:
```
NOTICE: Skipping optional dependency: espressif/esp_hosted
# repeated about 30 times
CMake Error at $IDF_PATH/tools/cmake/build.cmake:610 (message):
  ERROR: Because project depends on espressif/esp_video (0.*) which doesn't
  match any versions, version solving failed.

[... 113 lines cut ...]
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
NOTICE: Skipping optional dependency: espressif/esp_hosted
CMake Error at /workspace/esp_dev/esp-idf/tools/cmake/build.cmake:610 (message):
  ERROR: Because project depends on espressif/esp_video (0.*) which doesn't
  match any versions, version solving failed.

Call Stack (most recent call first):
  /workspace/esp_dev/esp-idf/tools/cmake/project.cmake:717 (idf_build_process)
  CMakeLists.txt:6 (project)

....-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /workspace/esp_dev/esp-idf/simple_video_server/build/log/idf_py_stderr_output_83880 and /workspace/esp_dev/esp-idf/simple_video_server/build/log/idf_py_stdout_output_83880
```

nopnop2002, 2025-04-16:
>-- IDF_TARGET not set, using default target: esp32            

This is for esp32p4 only.

truck-iguana, 2025-04-16:
@nopnop2002 I tried looking all over and somehow I missed that, right above a command I referenced too 🤦 
Sorry for this.