espressif/esp-idf #15800: ESP32-S3:BLE Crashed when disconnect

xianlezheng, 2025-04-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.3

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3 Chip

### Power Supply used.

USB

### What is the expected behavior?

Use V5.3.3 version, BluFi: BLUFI ble disconnect and the system crashed as the follow stack:

```

2025-04-16 22:10:35.207 I (280986) BluFi: BLUFI ble disconnect

assert failed: heap_caps_free heap_caps_base.c:74 (heap != NULL && "free() target pointer is outside heap areas")

Backtrace: 0x4037613a:0x3fcf3b30 0x40384109:0x3fcf3b50 0x4038c4b1:0x3fcf3b70 0x40376b46:0x3fcf3c90 0x4038c509:0x3fcf3cb0 0x4000a8b5:0x3fcf3cd0 0x4000770e:0x3fcf3cf0 0x4204982d:0x3fcf3d10 0x4204d309:0x3fcf3d50 0x40028f01:0x3fcf3d70 0x420527d3:0x3fcf3d90 0x4000d025:0x3fcf3dc0 0x4002c4a5:0x3fcf3de0 0x4204459b:0x3fcf3e00 0x403780bd:0x3fcf3e20 0x4037826f:0x3fcf3e40
--- 0x4037613a: panic_abort at /Users/NoPanic/esp/v5.3.3/esp-idf/components/esp_system/panic.c:463
--- 0x40384109: esp_system_abort at /Users/NoPanic/esp/v5.3.3/esp-idf/components/esp_system/port/esp_system_chip.c:87
--- 0x4038c4b1: __assert_func at /Users/NoPanic/esp/v5.3.3/esp-idf/components/newlib/assert.c:80
--- 0x40376b46: heap_caps_free at /Users/NoPanic/esp/v5.3.3/esp-idf/components/heap/heap_caps_base.c:74 (discriminator 1)
--- 0x4038c509: free at /Users/NoPanic/esp/v5.3.3/esp-idf/components/newlib/heap.c:39
--- 0x4000a8b5: r_emi_free_em_mapping_by_offset in ROM
--- 0x4000770e: r_ble_util_buf_adv_tx_free in ROM
--- 0x4204982d: hci_le_set_adv_data_cmd_handler at ??:?
--- 0x4204d309: r_llm_hci_command_handler at ??:?
--- 0x40028f01: llm_hci_command_handler_wrapper in ROM
--- 0x420527d3: r_ke_task_schedule at ??:?
--- 0x4000d025: r_ke_event_schedule in ROM
--- 0x4002c4a5: r_rwip_schedule in ROM
--- 0x4204459b: rw_schedule at ??:?
--- 0x403780bd: ble_try_turn_on_pll_track at ??:?
--- 0x4037826f: btdm_controller_task at ??:?
```

### What is the actual behavior?

The system crashed as follow log.

```
ELF file SHA256: 7a7e8aeb5

[... 113 lines cut ...]
Saved PC:0x40376098
--- 0x40376098: esp_restart_noos at /Users/NoPanic/esp/v5.3.3/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:158

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2820,len:0x1ad0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xf10
load:0x403cb700,len:0x3518
entry 0x403c89c0
I (31) boot: ESP-IDF v5.3.3 2nd stage bootloader
I (31) boot: compile time Apr 16 2025 21:21:45
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.1
I (38) boot: efuse block revision: v1.2
I (43) qio_mode: Enabling QIO for flash chip MXIC
I (48) boot.esp32s3: Boot SPI Speed : 80MHz
I (53) boot.esp32s3: SPI Mode       : QIO
I (58) boot.esp32s3: SPI Flash Size : 8MB
I (62) boot: Enabling RNG early entropy source...
```

### Diagnostic report archive.

[diag-c47e9174-5ca3-4b62-bca4-c1aac6d11466.zip](https://github.com/user-attachments/files/19779444/diag-c47e9174-5ca3-4b62-bca4-c1aac6d11466.zip)

esp-zhp, 2025-05-19:
Hello,

Thank you for reporting this issue.

We have identified and fixed the BLE disconnection crash on ESP32-S3. The fix has already been merged into the latest `master` branch (commit `4602665e5d9b94cef26f1992725db329e334d1c4`) as well as the stable `v5.3` branch (commit `3dbcb7b7f8c5f3157232da2cdae3e7f5399699e4`).

We recommend upgrading your project to the latest version of the ESP-IDF to benefit from this fix and other improvements. Please avoid applying the fix manually via cherry-pick, and instead use the official updated releases or branches.

If you continue to experience any issues after upgrading, please let us know.

esp-zhp, 2025-05-19:
4602665e5d9b94cef26f1992725db329e334d1c4