espressif/esp-idf #15801: async_handlers example not working after first async request

mattcarter11, 2025-04-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### IDF version.
v5.4.1

### Espressif SoC revision.
ESP32-D0WD (revision v1.1)

### Operating System used.
Windows

### How did you build your project?
With the VSCode esp-idf extension

### If you are using Windows, please specify command line type.
None

### Development Kit.
FireBeatle

### Power Supply used.
USB

### What is the expected behavior?
To be able to do the same async request more than once

### What is the actual behavior?
1. The first async response works fine
2. The consequent async responses don't work, they don't even reach the long_async_handler.

### Steps to reproduce.
1. copy async_handlers example to a new folder.
2. build example.(I shortened the duration in long_async() to avoid wasting time).
3. for (int i = 0; i < 10; i++) {
4. flash and run.
5. Accessing the 'long handler' link in the browser (or use curl).
6. Accessing the 'long handler' link in the browser again (or use curl). This second request wont work, getting stuck

### Debug Logs.

```
I (4475) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.173
I (4535) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:26dc:c3ff:fe94:63a0, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4535) example_common: Connected to example_netif_sta
I (4545) example_common: - IPv4 address: 192.168.1.173,
I (4545) example_common: - IPv6 address: fe80:0000:0000:0000:26dc:c3ff:fe94:63a0, type: ESP_IP6_ADDR_IS_LINK_LOCAL
I (4555) example: starting async req task worker
I (4555) example: starting async req task worker
I (4565) example: Starting server on port: '80'
I (4575) example: Registering URI handlers
I (4575) main_task: Returned from app_main()
I (53635) example: uri: /long
I (53645) example: invoking /long
I (53645) example: uri: /long
```

[image]

### More information
addint `httpd_resp_set_hdr(req, "Connection", "close");` on the last chuck response closes the connection and everithing works fine. The wierd thing is that in theory `HTTPD_DEFAULT_CONFIG` sets `keep_alive_enable = false`

hrushikesh430, 2025-04-21:
Hello @mattcarter11, I tried to reproduce your issue but unable to reproduce, on my side it is working fine. Did you changed any configuration or app?

Alvin1Zhang, 2025-04-30:
@mattcarter11 Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

hrushikesh430, 2025-05-06:
Hi @mattcarter11, I am closing this issue, due to no response. Please feel free to reopen the issue for further queries. Thank you.

closed 2025-05-06 by hrushikesh430