espressif/esp-idf #15804: ESP-IDF INSTALLATION

MuhdBasheerKt, 2025-04-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

**After installing the esp-idf-tools-setup-espressif-ide-3.1.0-with-esp-idf-5.3.1 setup file, the installation was completed. However, an error occurred while setting the tools into the IDF-PATH. Please refer to the logs below for more details.**

### What is the actual behavior?

IDE is opening, but can't create any project

### Steps to reproduce.

1. Reinstalled the esp-idf
2. run install.bat again after getting the error, but still facing same issue

...

### Build or installation Logs.

```plain
Using Python in C:\Espressif\python_env\idf5.3_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.3.1

Adding ESP-IDF tools to PATH...
WARNING: tool xtensa-esp-elf-gdb version 14.2_20240403 is installed, but has reported version unknown
ERROR: tool xtensa-esp-elf-gdb has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v5.3.1\install.bat' to install it.
WARNING: tool riscv32-esp-elf-gdb version 14.2_20240403 is installed, but has reported version unknown
ERROR: tool riscv32-esp-elf-gdb has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v5.3.1\install.bat' to install it.
WARNING: tool xtensa-esp-elf version esp-13.2.0_20240530 is installed, but has reported version unknown
ERROR: tool xtensa-esp-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v5.3.1\install.bat' to install it.
WARNING: tool riscv32-esp-elf version esp-13.2.0_20240530 is installed, but has reported version unknown
ERROR: tool riscv32-esp-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v5.3.1\install.bat' to install it.
WARNING: tool esp32ulp-elf version 2.38_20240113 is installed, but has reported version unknown
ERROR: tool esp32ulp-elf has no installed versions. Please run 'C:\Espressif\frameworks\esp-idf-v5.3.1\install.bat' to install it.
```

mfialaf, 2025-04-17:
Hi @MuhdBasheerKt 

May I ask what you mean by "reinstalled esp-idf"? Did you re-run the IDE Installer (with default setting or with any changes?), or did you use the install.bat script manually? 

Could you please describe step-by-step what you're doing in the installer, and at which point the error occurs? I'd like to try to reproduce the issue. I tested the IDE installer in a clean environment and was able to build an example project without issues.

During installation, the system environment variable IDF_TOOLS_PATH is set, which points to the directory where all toolchains are downloaded and installed (by default `C:\Espressif`). The issue might be that the toolchains are either missing or don't have executable permissions.

Could you please check the value of the IDF_TOOLS_PATH variable and share the log output from re-running the `install.bat` script?

MuhdBasheerKt, 2025-04-23:
HI @mfialaf ,
I uninstalled ESP-IDF and then reinstalled it, but I encountered the same issue.

After that, I tried running the install.bat script manually, followed by the export.bat script, but the issue persisted.

The problem occurs after the installation of ESP-IDF. When I try to open the ESP-IDF Command Prompt or PowerShell, I see the same error I mentioned in my initial post.

I also attempted to set the environment variables manually, but unfortunately, that didn’t resolve the issue either.

Please find the log output and a screenshot of the error below for your reference.

[IDF_TOOLS_PATH not set. Setting to.txt](https://github.com/user-attachments/files/19860532/IDF_TOOLS_PATH.not.set.Setting.to.txt)
[image]

mfialaf, 2025-04-24:
Thank you @MuhdBasheerKt for the detailed logs.

I was investigating the issue and I am still not sure what causes this. It seems that the failing tools are downloaded, extracted but then either somehow broken, or maybe there are read/execute permission issues. It's also interesting that some of the tools succeeds (like `Skipping cmake@3.24.0 (already installed)`) but others fails.

Could you please try these few things and report back to me how it went?

- Based on the log you provided executing the commands from the destination directory fails. Lets try some testing on the failing tool `xtensa-esp-elf`. Open the ESP-IDF PowerShell and:
  1)  Run the tool with the exact directory as `C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp-elf-gcc --version` or `C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin\xtensa-esp-elf-gcc.exe --version`
  2) Try to invoke the tool like `xtensa-esp-elf-gcc --version` or `xtensa-esp-elf-gcc.exe --version`.    If this succeeds, please run command `Get-Command xtensa-esp-elf-gcc` and verify that the source path is same as the one from previous step.

- Does it make a difference if you run the ESP-IDF PowerShell as Administrator? If so, than it seems there is read/execute permission issue.

- You could try to remove the whole `C:\Espressif` directory. Then open the ESP-IDF PowerShell and/or run the install script. This triggers the whole install process from scratch, meaning all the tools are going to be downloaded and installed again.

dobairoland, 2025-05-05:
I'm closing this for inactivity and insufficient information for reproducing it.

closed 2025-05-05 by dobairoland