espressif/esp-idf #15806: troubles with LEDC initialization

ravencarcass, 2025-04-17:
Hello, 
I am having some problems with LEDC initialization for a project.
I am using:
 - ESP32C6 
 - ESP-IDF: V5.4

While building, it did not report any error. 
But when I try to initialize the LEDC function, it reports the following error in the terminal:
```
E (621) ledc: ledc_timer_del(736): LEDC is not initialized
E (631) LEDC: failed to configure timer: ESP_ERR_INVALID_STATE
```
The initialization code: 
```
    ledc_timer_config_t timer;
        timer.speed_mode         = LEDC_MODE;
        timer.timer_num          = LEDC_TIMER;
        timer.freq_hz            = LEDC_FREQUENCY;
        timer.duty_resolution    = LEDC_TIMER_8_BIT;
        timer.clk_cfg            = LEDC_AUTO_CLK;
    esp_err_t retT = ledc_timer_config(&timer);

    if (retT != ESP_OK) {
        ESP_LOGE("LEDC", "Failed to configure timer: %s", esp_err_to_name(retT));
        return;
    }
   
    ledc_channel_config_t channel;
        channel.gpio_num       = LEDC_OUTPUT_IO;
        channel.speed_mode     = LEDC_MODE;
        channel.channel        = LEDC_CHANNEL;
        channel.intr_type      = LEDC_INTR_DISABLE;
        channel.timer_sel      = LEDC_TIMER;
        channel.duty           = LEDC_DUTY;
        channel.hpoint         = 0;
        esp_err_t ret = ledc_channel_config(&channel);

    if (ret != ESP_OK) {
        ESP_LOGE("LEDC", "Failed to configure channel: %s", esp_err_to_name(ret));
        return;
    }
```
I have already tried to fix it with a {0}, but to no avail. 
I appreciate your efforts.

igrr, 2025-04-17:
ledc_timer_del will be called only if ledc_timer_config_t::deconfigure field is set.

Could you please change the code to 
```
    ledc_timer_config_t timer;
    memset(&timer, 0, sizeof(timer));
      // the rest is the same
```

and try again?

ravencarcass, 2025-04-17:
I'm sorry I found the error. 
The timer initialization was not in the right order and changed it to this 
```
        timer.speed_mode         = LEDC_MODE;
        timer.timer_num          = LEDC_TIMER;
        timer.freq_hz            = LEDC_FREQUENCY;
        timer.duty_resolution    = LEDC_TIMER_8_BIT;
        timer.clk_cfg            = LEDC_AUTO_CLK;
```

 I changed the clk_cfg to `LEDC_USE_XTAL_CLK.` 
This solved the problem.

closed 2025-04-17 by ravencarcass