espressif/esp-idf #15807: Is there support for using ST77916 with i80?

SonOfPear93, 2025-04-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've recently bought a display that supposedly uses a ST77916 display driver, and the pinout suggests it will only work using the i80 bus.

The readme for esp_lcd_ST77916 claims to only work with SPI & QSPI, does this mean I cannot use my display with esp-idf at the moment?

I have attempted to use the esp_lcd_ST77916 component anyway, and whilst it does not error, nothing is being displayed to the screen. (in the snippet below I am turning off the backlight, elsewhere I am turning it back on after initialising lvgl etc)

Here is a link to the display I have; https://a.aliexpress.com/_EzG0JQG

Here is a snippet of the code I am attempting to use;

```
    ESP_LOGI(TAG, "Initialize I80 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .dc_gpio_num = GPIO_NUM_2,
        .wr_gpio_num = GPIO_NUM_35,
        .clk_src	= LCD_CLK_SRC_DEFAULT,
        .data_gpio_nums =
        {
            GPIO_NUM_34,
            GPIO_NUM_7,
            GPIO_NUM_6,
            GPIO_NUM_38,
            GPIO_NUM_39,
            GPIO_NUM_40,
            GPIO_NUM_41,
            GPIO_NUM_42,
        },
        .bus_width = 8,
        .max_transfer_bytes = 360 * 360 * static_cast<int>(sizeof(uint16_t)),
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = GPIO_NUM_36,
        .pclk_hz = 6528000,
        .trans_queue_depth = 20,
        .lcd_cmd_bits = 8,
        .lcd_param_bits = 8,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
            .dc_data_level = 1,
        },
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));

    ESP_LOGI(TAG, "Install ST77916 panel driver");
    esp_lcd_panel_handle_t panel_handle = NULL;
    const esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = GPIO_NUM_1,
        .rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,     // Implemented by LCD command `36h`
        .bits_per_pixel = 16,    // Implemented by LCD command `3Ah` (16/18)
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_st77916(io_handle, &panel_config, &panel_handle));
    esp_lcd_panel_reset(panel_handle);
    esp_lcd_panel_init(panel_handle);
    esp_lcd_panel_disp_on_off(panel_handle, true);
```

SonOfPear93, 2025-04-17:
[ST77916_SPEC_V1.0_20220830(1)(2)(2)(1).pdf](https://github.com/user-attachments/files/19795601/ST77916_SPEC_V1.0_20220830.1.2.2.1.pdf)
[HD18001C22规格书.pdf](https://github.com/user-attachments/files/19795600/HD18001C22.pdf)

for reference, here are the datasheets for the display & driver

suda-morris, 2025-04-18:
Duplicated with https://github.com/espressif/esp-iot-solution/issues/498

Since the ST77916 device driver is hosted in that repo, let's continus the discussion there. I will close this one.

closed 2025-04-18 by suda-morris