espressif/esp-idf #15808: Documentation Rendering Issue: uart_driver_install() Parameter Description

OmarYarali, 2025-04-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Link to affected page:
https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32/api-reference/peripherals/uart.html

Issue:
The description for uart_driver_install() lists the parameters in this order:

```
- Size of TX ring buffer  
- Size of RX ring buffer
```
However, the actual function signature is:
```c
esp_err_t uart_driver_install(uart_port_t uart_num,
                              int rx_buffer_size,
                              int tx_buffer_size,
                              int queue_size,
                              QueueHandle_t *uart_queue,
                              int intr_alloc_flags);
```
This may confuse users, since the RX buffer size comes before the TX buffer size.

Notes:

I checked the source .rst file in the repo, and it appears to be correct.

This seems to be a rendering or content sync issue on the documentation website.

suda-morris, 2025-04-18:
Fixed in https://github.com/espressif/esp-idf/commit/c465b75ed60d447f5fe834308b0c3567eb8848c5 already.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#install-drivers

suda-morris, 2025-04-18:
Ah the backport is also finished: https://github.com/espressif/esp-idf/commit/24747868dd0a001e661b20d6ba1c67bfaad2951a

The programming guide get synced in the next v5.4 release, i.e. v5.4.2

closed 2025-04-18 by suda-morris