espressif/esp-idf #15810: Support more than one TinyUSB instance on ESP32-P4

maufl, 2025-04-18:
### Is your feature request related to a problem?

The ESP32-P4 has two USB interfaces (plus the USB Serial/JTAG controller, as far as I can tell). According to the documentation of the chip, you can choose different pins for USB D+ and D-, depending on which interface you use, meaning it should be possible to use both at the same time with different pins. As far as I can tell, the ESP-IDF and the TinyUSB component don't allow you to do this yet, because you can't specify the USB interface or pins you want to use.

### Describe the solution you'd like.

It would be great if you could choose the pins to use when initializing TinyUSB and have two TinyUSB instances running at the same time.

Alvin1Zhang, 2025-04-18:
Thanks for reporting.

closed 2025-04-22 by roma-jam

roma-jam, 2025-04-22:
@igrr 

TinyUSB limitation. 

Simple answer: there is only one internal static variable `_usbd_rhport` which will be initialized after first stack initialization and it prevents to initialize stack again, even if the peripheral has been changed.

A bit detailed answer: we don't have TinyUSB instance handle, so there are lots of internal variables which should be shared between instances running on two different peripherals (such as buffers for control transfer and so on). To be able to run it simultaneously on two (or more) peripherals,  we need at least distinguish these variables and have them for every instance of TinyUSB stack.

roma-jam, 2025-04-22:
I have created the feature request: https://github.com/hathach/tinyusb/issues/3092
In case someone will be looking for it.