espressif/esp-idf #15811: Cannot get data from firebase realtime database

yunlong12, 2025-04-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use esp32 chip, and my code is attached as 

[combined_files.txt](https://github.com/user-attachments/files/19811086/combined_files.txt)

my code is to send data out to and get data from Firebase Realtime database. 

When I use esp idf 5.3.1, I can do these operations succesfully, and, from the termial, I can see all the info from ESP32. And the info is attached as 

[info_with_espIDF_v5_3_1.txt](https://github.com/user-attachments/files/19811048/info_with_espIDF_v5_3_1.txt)

When I use esp idf 5.4.1, I fail and the error info is,

`W (31991) HTTP_CLIENT: Connection timed out before data was ready!
E (31991) FirebaseApp: Error while performing request esp_err_t code=0x7007 | status_code=-1
E (31991) FirebaseApp: request: url=https://tong-feng-kou-kong-zhi-default-rtdb.europe-west1.firebasedatabase.app/person3.json?auth=eyJhbGciOiJSUzI1NiIsImtpZCI6Ijg1NzA4MWNhOWNiYjM3YzIzNDk4ZGQzOTQzYmYzNzFhMDU4ODNkMjgiLCJ0eXAiOiJKV1QifQ.eyJpc3MiOiJodHRwczovL3NlY3VyZXRva2VuLmdvb2dsZS5jb20vdG9uZy1mZW5nLWtvdS1rb25nLXpoaSIsImF1ZCI6InRvbmctZmVuZy1rb3Uta29uZy16aGkiLCJhdXRoX3RpbWUiOjE3NDQ5Njg2MjYsInVzZXJfaWQiOiJnSTFZY0pSanZ6WUt0aGh0cEZINWxEWjZhakwyIiwic3ViIjoiZ0kxWWNKUmp2ellLdGhodHBGSDVsRFo2YWpMMiIsImlhdCI6MTc0NDk2ODYyOCwiZXhwIjoxNzQ0OTcyMjI4LCJlbWFpbCI6ImhlbGxvX3Rlc3RAZ21haWwuY29tIiwiZW1haWxfdmVyaWZpZWQiOmZhbHNlLCJmaXJlYmFzZSI6eyJpZGVudGl0aWVzIjp7ImVtYWlsIjpbImhlbGxvX3Rlc3RAZ21haWwuY29tIl19LCJzaWduX2luX3Byb3ZpZGVyIjoicGFzc3dvcmQifX0.uqXDlESoiUH3A4R3GlyvDl7A5Iz0OV8cC1lFLR42KBEARntpJN_VyTjsRgHRRBccTyVo9tifxLcYeNfySY2zInmbBd9j7SLA1Ej2eoP1lkx9n8kpadexkOHv127jtcEqluHGtkpUFZ-53YSs1Hp1Iokyp7ardLFZMfPYNhC2G_hqdc1j-yj0tkcNcNG3OHMVZvnpVw1KToyoij0S2KzZhaVcP7IbCPItsL1JMN2-LeliZK62-Z2gwkhD6aV6OGmAB5jb15xvEkBQqjjsXruzcs8y5khBZNq3St-QTV3OaRESWh9xQhVfdGy8aHWycQiX5wPu8t9fjpZei-MNvCGy3Q
method=0
post_field=
E (32091) FirebaseApp: response=

E (32091) RTDB: Failed to get data after refreshing token. double check account credentials or database rules `

the whole info of using esp idf v5.4.1 is also attached, 

[info_with_espIDF_v5_4_1.txt](https://github.com/user-attachments/files/19811061/info_with_espIDF_v5_4_1.txt)

Becuase, I use the same code, so I think the issue lies in the esp idf version change.

Thanks.

yunlong12, 2025-04-18:
I can send data to Firebase realtime database succesfully, with both the esp idf v 5.3.1 and v 5.4.1.

yunlong12, 2025-04-18:
the code project is uploaded here.

[ESP32_IDF_Firebase_info_exchange.zip](https://github.com/user-attachments/files/19811350/ESP32_IDF_Firebase_info_exchange.zip)

nileshkale123, 2025-04-21:
Hello @yunlong12 ,

Just to confirm, it seems you're able to successfully send data to Firebase, but you're encountering an issue when trying to retrieve data from it. Please let me know if that's correct so I can review the code. If the issue is already resolved, feel free to close it.

yunlong12, 2025-04-23:
Hi @nileshkale123 ,
Yes, that is it. The issue is still there. Now, I am still using v5.3.1.

nileshkale123, 2025-06-03:
Hello @yunlong12 ,

Please provide me the .zip or .tar file containing all files and the necessary steps to reproduce the issue, so we can debug it further.

You can also reproduce the issue in the example `esp_http_client_example` if possible, and provide me with the patch.

closed 2025-06-17 by nileshkale123