espressif/esp-idf #15813: Problem with large monolithic C code Result LoadProhibited fault

filzek, 2025-04-18:
@igrr Hi Ivan,

I think this goes to you as you are more ack on how the compiler and linker currently works in the esp-idf environment by 5.4.1.

We are developing a new solution and it was designed just to have de main file in the cmakelist.

The main file includes all other C files and do not even use headers, working as a huge unique monolithic block.

The compiler and linker works and produce the workable firmware.

A simple main app the just has the code

const esp_partition_t *partition = esp_partition_find_first(ESP_PARTITION_TYPE_DATA, ESP_PARTITION_SUBTYPE_ANY, "storage");

Blows with:

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40190a0f  PS      : 0x00060f30  A0      : 0x80190aa5  A1      : 0x3ffbb070
--- 0x40190a0f: esp_partition_next at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_partition/partition.c:331 (discriminator 1)

A2      : 0x3ffb6cb8  A3      : 0x00000001  A4      : 0x00000001  A5      : 0x000000ff
A6      : 0x8024337a  A7      : 0x3ffb6cb8  A8      : 0x00000000  A9      : 0x3ffbb060
A10     : 0xffffff74  A11     : 0x3ffbab0c  A12     : 0x000000ff  A13     : 0x00cacec2
A14     : 0x00000001  A15     : 0x40404040  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c
EXCVADDR: 0x8024337e  LBEG    : 0x400954d1  LEND    : 0x400954f5  LCOUNT  : 0x80190a35

0x80190a35
--- 0x400954d1: strcmp at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:163
--- 0x400954f5: strcmp at /builds/idf/crosstool-NG/.build/xtensa-esp-elf/src/newlib/newlib/libc/machine/xtensa/strcmp.S:181

Backtrace: 0x40190a0c:0x3ffbb070 0x40190aa2:0x3ffbb090 0x40190ab7:0x3ffbb0b0 0x400f9241:0x3ffbb0d0 0x401171a9:0x3ffbb960 0x4023cde2:0x3ffbbaf0 0x40099ba2:0x3
--- 0x40190a0c: esp_partition_next at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_partition/partition.c:331
--- 0x40190aa2: esp_partition_find at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_partition/partition.c:315
--- 0x40190ab7: esp_partition_find_first at C:/Espressif/frameworks/esp-idf-v5.4/components/esp_partition/partition.c:356

So just there code and data in the firmware and the app main just call one line there is nothing more there.

We know that the partition is perfect as if we upload just the app with the basic headers and no includes at all to the device it runs.

So, we think somehow the linker and compiler are somehow has misalignment  in all sectors and the memory/heap are already corrupted on boot.

We think this could be a bug or even a CVM on the building system.

igrr, 2025-04-22:
This seems to be a run-time memory corruption. The code mentioned in the backtrace is as follows:

https://github.com/espressif/esp-idf/blob/4c2820d377d1375e787bcef612f0c32c1427d183/components/esp_partition/partition.c#L330-L331

In this case, "discriminator 1" refers to `it->type` dereference operator. LoadProhibited exception indicates that the CPU has attempted to perform an access to an invalid address ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#loadprohibited-storeprohibited)). We can see the address in the exception dump (EXCVADDR register): 0x8024337e. This is indeed not a valid RAM address on an ESP chip. We can therefore conclude that the pointer `it` was `0x8024337e` minus the offset of `type` field in the structure. So, `it` was an invalid pointer.

Since the linked list of partition structures is constructed at start up and never modified afterwards, it's reasonable to conclude that the contents of the structure was damaged at run time, overwritten by some other code. Debugging heap corruption issues is indeed complex and time-consuming, but we do have a few suggestions [in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html#finding-heap-corruption).

igrr, 2025-04-22:
Hard to say, I would expect that if globals are declared but not referenced in the code, they should be all removed at link time. Perhaps there is some other factor which might be causing a memory corruption.

I believe you have a mail thread with our support team, if you can send the project there, that would be great. Thanks in advance.

filzek, 2025-07-01:
@Alvin1Zhang monolithic coding corrupts everywhere in the bss, not sure why this happen.

We did refactor everything to modern C, the problem still happens with some parts as well, but, seems that issue triggered by some undocumented or old not fixed sdkconfig update, realted to dhcp settings.

The corruption on BSS do not show without monolithic code.