espressif/esp-idf #15814: httpd must urldecode incoming urls from HTTP

codewitch-honey-crisis, 2025-04-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm not sure if I should have filed this under bug fix or feature request, but I'm leaning on the former and here's why:

Your server is not very robust in the face of modern browsers. Consider the url `http://www.foo.com/bar%20baz%20boo` or `http://www.foo.com/bar+baz+boo`

That's a valid url, and it may be represented several different ways depending on how the browser encodes (rfc tables vs html5 tables, mainly "%20" vs "+" for spaces)

Now the problem with that is that in order to handle all cases robustly I need to register multiple URL handlers for the same URL!

If you're implementing the HTTP spec properly your server should be handling the decoding of the URL paths before handing it off to your request handlers and you're not doing that. I ran face first into this problem on MY FIRST webserver project, so this isn't niche.

20162026, 2025-04-20:
1) `+` is only treated as a space in form data or a query string. If you found a http server that threats both /bar+baz+boo and /bar%20baz%20boo as the same url, you should fill a bug report

2) /bar and /bar/ are different urls, some web servers might automatically redirect or drop/add trailing slash but thats certainly not a standardized behaviour

codewitch-honey-crisis, 2025-04-21:
Which makes it unfortunate that you even have this facility in the first place, rather than just giving the developer the HTTP(S) request and letting them decide which handler to use. Oh well. I guess I'll just register half a dozen handlers for 2-3 pages.

igrr, 2025-04-21:
There is an "escape hatch" for custom URI matching in the form of httpd_config_t::uri_match_fn. By default, it is NULL and the webserver uses the default URI matching behavior. You can pass either httpd_uri_match_wildcard helper or your own function, instead.

codewitch-honey-crisis, 2025-04-21:
That's amazing thank you! I must have missed it in the documentation. That takes care of my issue at least. Have a great day.

closed 2025-04-21 by codewitch-honey-crisis