espressif/esp-idf #15816: SPI Use MOSI for both RX and TX with Quad Output

salivo, 2025-04-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I’m working with a display that uses Quad SPI (4-line) communication.
Writing to the display works well — I'm able to transmit pixel data using Quad Output, DMA, and Half-Duplex mode successfully.

However, when trying to implement read functionality, I encountered a limitation:
There is currently no way to read from the MOSI (IO0) pin — which the display uses for returning data.
This makes it impossible to receive responses from the display using standard SPI transactions.

I have tried to manually generate the clock, and it works, but I don't think that is the best practice.
Is there an intended method or workaround for this setup? I’d appreciate clarification.

Let me know if you'd like to add logs, hardware model, or specific register behavior to make it even more solid.

wanckl, 2025-04-21:
@salivo  Usually read from LCD has frequncy limit which more slow then wirte to it, Although SPI is now able to change the `device_handle`'s frequncy,  but if you are using different pin, you can add another device for read.    
@Lzw655 WDYT

If read can only from MOSI line, you may need add read_dev as `SPI_DEVICE_3WIRE`

salivo, 2025-04-21:
Yes, I also thought about this, but this solution has one problem: how to keep the CS pin active?
I’ve tried this, but as the code runs, the second transmit must wait for the bus to be released—however, it only gets released after transmitting, so it waits with `portMAX_DELAY`.
```c
esp_err_t display_receive_spi(uint8_t addr, uint8_t* data, size_t data_len) {
    spi_device_acquire_bus(spi_lcd_handle, portMAX_DELAY);
    spi_transaction_t t = {
        .cmd = CMD_WRITE,
        .addr = addr,
        .length = 0,
        .tx_buffer = NULL,
        .flags = SPI_TRANS_CS_KEEP_ACTIVE
    };
    spi_device_polling_transmit(spi_lcd_handle, &t);
    spi_transaction_t t_recv = {
        .cmd = CMD_READ,
        .addr = addr,
        .rxlength = data_len * 8,
        .rx_buffer = data
    };
    spi_device_polling_transmit(spi_lcd_read_handle, &t_recv);
    spi_device_release_bus(spi_lcd_handle);
    return ESP_OK;
}
```

salivo, 2025-04-21:
Ahh, I figured out how to keep the CS pin active — I just need to perform a single transaction:
```c
esp_err_t display_receive_spi(uint8_t addr, uint8_t* data, size_t data_len) {
    spi_transaction_t t = {
        .cmd = CMD_READ,
        .addr = addr,
        .rx_buffer = data,
        .rxlength = data_len * 8,
    };
    return spi_device_polling_transmit(spi_lcd_read_handle, &t);
}
```

But I ran into another issue: when I add `spi_lcd_read_handle`, like this:
```c
esp_err_t add_spi_CO5300_dev() {
    spi_device_interface_config_t lcd_interface_conf = {
        .spics_io_num = AMOLED_SPI_CS,
        .clock_speed_hz = SPI_MASTER_FREQ_80M,
        .mode = 0,
        .queue_size = 1,
        .command_bits = 8,
        .dummy_bits = 8,
        .address_bits = 16,
        .flags = SPI_DEVICE_HALFDUPLEX
    };

    spi_device_interface_config_t lcd_interface_read_conf = {
        .spics_io_num = AMOLED_SPI_CS,
        .clock_speed_hz = SPI_MASTER_FREQ_20M,
        .mode = 0,
        .queue_size = 1,
        .command_bits = 8,
        .dummy_bits = 8,
        .address_bits = 16,
        .flags = SPI_DEVICE_HALFDUPLEX | SPI_DEVICE_3WIRE
    };

    ESP_RETURN_ON_ERROR(spi_bus_add_device(DISPLAY_SPI_HOST, &lcd_interface_conf, &spi_lcd_handle), TAG, "failed to add SPI device");
    return spi_bus_add_device(DISPLAY_SPI_HOST, &lcd_interface_read_conf, &spi_lcd_read_handle);
}
```
...it doesn't send data only receive. Maybe adding the second device changes the behavior of some pins.

wanckl, 2025-04-21:
@salivo You need manually control CS pin by add both device as `cs_io_num=-1` then write_gpio through 2 devices' `pre/post_trans_cb`

salivo, 2025-04-22:
Yes, this works perfectly. Thanks so much for the help!

closed 2025-04-22 by salivo