espressif/esp-idf #15817: esp32p4 + espidf5.4 + espadf master enable psram ecc  error

logicDance-mcu, 2025-04-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Espressif SoC revision.

ESP32P4-V1.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

no

### Power Supply used.

USB

### What is the expected behavior?

It can run normally after ecc is turned on

### What is the actual behavior?

[enable ecc log.txt](https://github.com/user-attachments/files/19820467/enable.ecc.log.txt)

[disable ecc log.txt](https://github.com/user-attachments/files/19820474/disable.ecc.log.txt)

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

Icarus113, 2025-04-21:
It's hard to tell the cause of this issue from the posted log. Would you help debug further to see where the program gets stuck? BTW, you can try enabling `CONFIG_ESP_PANIC_HANDLER_IRAM` to see if there're more info.

ginkgm, 2025-05-23:
The log shows that, there is something wrong that causes the panic, and due to some reason PANIC handler also can't work when it is in the PSRAM. We can't get any further information from the log aside from that.

Please enable the CONFIG_ESP_PANIC_HANDLER_IRAM and try again.

Feel free to reopen the issue when there is further information.

closed 2025-05-23 by ginkgm