espressif/esp-idf #15819: Could `esp_ota_begin` accept factory partitions?

pvginkel, 2025-04-19:
### Is your feature request related to a problem?

I'm trying to optimize flash usage when using OTA. I want to do this by having just one OTA partition. Instead of alternating between `ota_0` and `ota_1`, I want to alternate between `factory` and `ota_0`. I realize that I then can't do a factory reset anymore, but that is acceptable for me.

I can easily implement the logic myself to detect the right partition. However, `esp_ota_begin` [refuses to take a partition with subtype `ESP_PARTITION_SUBTYPE_APP_FACTORY`](https://github.com/espressif/esp-idf/blob/d930a386dae78cfab75f313af3df67921e748fc4/components/app_update/esp_ota_ops.c#L67). I would like this check to be relaxed to allow me to use the `factory` partition as an OTA target partition.

### Describe the solution you'd like.

I would like `esp_ota_begin` to accept partitions with the subtype set to `ESP_PARTITION_SUBTYPE_APP_FACTORY`.

### Describe alternatives you've considered.

Hacking around this limitation using the following code:

```cpp
const auto update_partition = ota_config.update_partition;

const auto update_partition_subtype = update_partition->subtype;
if (update_partition->subtype == ESP_PARTITION_SUBTYPE_APP_FACTORY) {
    ((esp_partition_t *)update_partition)->subtype =
        ESP_PARTITION_SUBTYPE_APP_OTA_0;
}

const auto err = esp_ota_begin(
    update_partition,
    OTA_WITH_SEQUENTIAL_WRITES,
    &update_handle);

((esp_partition_t *)update_partition)->subtype = update_partition_subtype;

ESP_ERROR_CHECK(err);
```

kriegste, 2025-04-19:
Updating factory partitions using the regular OTA APIs would be a great feature!

chegewara, 2025-04-19:
That's the whole point of the factory partition, to not change it after production, ever.
If you need/want to have option to update factory partition, just make it OTA partition instead. Factory partition is optional.

chegewara, 2025-04-19:
Yet again, you can make it OTA type or just use non-ota API to update factory partition.

kriegste, 2025-04-19:
Making it OTA type requires changing the partition table. Last time I checked, there was no API to do this. And using non-OTA flash writing APIs is possible, yes, but requires a proper implementation which could be error-prone in itself. An official way- properly implemented- would be the safest option.

mbratch, 2025-04-19:
I don't have a "factory partition". I just have two OTA partitions. When the product is first flashed, it uses ota1.

```
nvs,      data, nvs,     0xa000,  0x3000,
otadata,  data, ota,     0xd000,  0x2000,
phy_init, data, phy,     0xf000,  0x1000,
ota_0,    app,  ota_0,   0x10000, 0x1e0000,
ota_1,    app,  ota_1,   0x1f0000,0x1e0000,
www,      data, spiffs,  0x3D0000,0x30000
```

pvginkel, 2025-04-20:
Thank you for the steer. I've verified this works and it's a perfect solution to the problem.

closed 2025-04-20 by pvginkel