espressif/esp-idf #15824: esp-wifi-mesh on esp32c5 does not establish wifi connections to children as a root node

naticklamb, 2025-04-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-2375-g1f46216a72

### Espressif SoC revision.

esp32-c5-wroom-1 v1.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-c5-wroom-1 v1.2

### Power Supply used.

USB

### What is the expected behavior?

I expect a esp-wifi-mesh child nodes to be able to communicate with root nodes.

### What is the actual behavior?

Instead, when the root node is running on an esp32c5, it does not communicate.  The root node's softAP does see initial contact enough to indicate MESH_EVENT_CHILD_CONNECTED, but does not follow through with its DHCP server assigning an IP address to the child node.
The child node is then stuck in this timeout state until the root node is powered off.  esp-wifi-mesh on the child node then faithfully becomes root itself.
See "== start of bug ==" in the debug log files that follow.

### Steps to reproduce.

cd .../esp-idf/examples/mesh/ip_internal_network
idf.py --preview set-target esp32c5
idf.py menuconfig  (or to whatever your wifi setup is)
 Example Configuration->channel 0, Router SSID "xxpass2", Router password "", Mesh AP Authentication Mode WIFI_AUTH_OPEN, Mesh\
 AP Password ""
idf.py build
idf.py -p /dev/ttyACM3 flash
screen /dev/ttyACM3
Let this esp-wifi-mesh instance become root.  Then startup a copy on another esp32 device.  It will try to connect to the C5 but fail. 
If you do the same experiment with a non-C5 as root node, it will work fine (i have tried with C3, S2, and plain with no problems).
I suspect the line:
  I (328223) mesh: [tx-wifi]tx_state:0x300f
indicates a compatibility problem with the esp-wifi-mesh package and the C5 WiFi stack.

### Debug Logs.

```plain
========== ON ESP32C5 =====================
ESP-ROM:esp32c5-eco1-20240726
Build:Jul 26 2024
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPI mode:DIO, clock div:1
load:0x40855ab0,len:0x176c
load:0x4084bfa0,len:0xd08
[... 420 lines cut ...]
E (80667) esp-tls: [sock=54] select() timeout
E (80667) transport_base: Failed to open a new connection: 32774
E (80667) mqtt_client: Error transport connect
I (80667) mesh_mqtt: MQTT_EVENT_ERROR
I (80667) mesh_mqtt: MQTT_EVENT_DISCONNECTED
I (80677) mesh_mqtt: sent publish returned msg_id=58799
============================= start of bug ==============================
I (80677) mesh_main: Sending routing table to [0] 48:31:b7:3e:57:00: sent with err code: 0
I (80687) mesh_main: Received Routing table [0] 48:31:b7:3e:57:00
I (80687) mesh_main: Received Routing table [1] 60:55:f9:fa:db:60
I (80687) mesh_main: Sending routing table to [1] 60:55:f9:fa:db:60: sent with err code: 0
I (82707) mesh_main: Tried to publish layer:1 IP:192.168.15.124
I (82707) mesh_mqtt: sent publish returned msg_id=16998
============================================================================
I (82707) mesh_main: Sending routing table to [0] 48:31:b7:3e:57:00: sent with err code: 0
I (82707) mesh_main: Sending routing table to [1] 60:55:f9:fa:db:60: sent with err code: 0
I (82717) mesh_main: Received Routing table [0] 48:31:b7:3e:57:00
I (82727) mesh_main: Received Routing table [1] 60:55:f9:fa:db:60
I (84717) mesh_main: Tried to publish layer:1 IP:192.168.15.124
I (84717) mesh_mqtt: sent publish returned msg_id=21716
I (84717) mesh_main: Sending routing table to [0] 48:31:b7:3e:57:00: sent with err code: 0
I (84717) mesh_main: Received Routing table [0] 48:31:b7:3e:57:00
I (84727) mesh_main: Received Routing table [1] 60:55:f9:fa:db:60
...
```

zhangyanjiaoesp, 2025-04-21:
@naticklamb thanks for reporting, we will check this issue

zhangyanjiaoesp, 2025-04-28:
@naticklamb After commenting out the following functions in the C5 ld file, the connection succeeds.

[image]

naticklamb, 2025-04-29:
Yes.  This fixes the C5 esp-wifi-mesh root connectivity problem.  Thank you so much for fixing this and responding promptly!!

closed 2025-04-30 by naticklamb

Lucia-Rubio-E, 2025-05-22:
@naticklamb  @zhangyanjiaoesp  @espressif-bot  @methodmissing  @svenfuchs 
I'm currently looking for an ESP32-C5 module or DevKit that comes with the ECO2 silicon version (revision v1.2).
Does anyone know a reliable place to buy a confirmed ECO2 version?
It can be a module (WROOM) or a full DevKit.
Any suggestions—official distributors, AliExpress links, or vendors who have confirmed stock—would be very helpful.

Thanks in advance!

naticklamb, 2025-06-18:
Ive been getting V1.3 ESP32-C5-WROOM-1 modules off Alibaba.
However I have noticed V1.3 does not start the USB port on boot or even in boot mode (gpio28 gnd, rst).  I Must have an external serial connection to flash (of course later fw updates are just with my own s/w).  This adds a cost to initial provisioning of our product.