espressif/esp-idf #15825: The LICENSE file in esp-idf and in some of espressif repositories doesn't have the correct Copyright

HalfSweet, 2025-04-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

A typical example, in
https://github.com/espressif/esp-idf/blob/master/LICENSE#L190
does not have the correct year and owner's name.

igrr, 2025-04-22:
Hi @HalfSweet, please check if this comment answers your concern: https://github.com/espressif/esp-idf/issues/9470#issuecomment-1202120785

HalfSweet, 2025-04-22:
Thanks for the reply, that solves the problem

closed 2025-04-22 by HalfSweet