espressif/esp-idf #15826: ULP - RTC GPIO missing functionality and documentation

jath03, 2025-04-21:
### Is your feature request related to a problem?

A fan I'm driving using the LP Core on an ESP32C6 was turning off  when the main CPU was in deep sleep mode.  I had already noticed that there were many `rtcio_ll_...` functions that weren't directly exposed to userspace or documented, but it wasn't until I dug through the register list in the technical reference manual that I found the register that configures the gpio drive strength during sleep, which didn't even have an `rtcio_ll_...` function written yet.  

### Describe the solution you'd like.

Improve RTC GPIO feature coverage in the ULP and main core along with documentation.  The fact that there is a separate drive strength value during sleep to configure isn't mentioned anywhere outside of the register details in the technical reference manual.

### Describe alternatives you've considered.

I have implemented a fix for my own issue, but do not have time to write a PR myself right now.

o-marshmallow, 2025-04-21:
Hello @jath03 ,

Which version of IDF are you using? Which RTC GPIOs are you using?

As documented here, https://docs.espressif.com/projects/esp-idf/en/stable/esp32c6/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories, it is possible to configure the RTC domain (peripherals and so RTC GPIOs) to stay on even during deep sleep, as such, you don't need to call `ll` functions, which are private. Try the following:

```
    /* Configure your RTC GPIOs */
    esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
   
    /* Load the ULP program and run it, it can use any RTC GPIO */

    /* Enter deep sleep, the ULP should still be able to drive the GPIOs */
    esp_deep_sleep_start();
```

jath03, 2025-04-21:
Hi, to be clear, my personal code is now functioning correctly and I know how to use deep sleep.  I'm just disappointed in the fact that I had to use `ll` functions and read through a TRM instead of being able to rely on the normally amazing documentation and software support.  

For instance only a limited subset of the [`ll` LP GPIO functions](https://github.com/espressif/esp-idf/blob/master/components/hal/esp32c6/include/hal/rtc_io_ll.h) are [available from the LP core](https://github.com/espressif/esp-idf/blob/master/components/ulp/lp_core/lp_core/include/ulp_lp_core_gpio.h).  Even comparing the [RTC GPIO docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/gpio.html#api-reference-rtc-gpio) and the [LP GPIO docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/ulp-lp-core.html#id16) shows the clear feature differences.  To run any pin hold functions from the LP core, I had to use `ll` functions.

Beyond these differences, there was not a single mention of a way to set the sleep mode drive strength for a pin anywhere in any of the GPIO documentation, nor is there any function to do so.  In the `LP_IO_GPIOn_REG` there is a field `LP_GPIO_GPIOn_MCU_DRV` (page 290 of the ESP32C6 Technical Reference Manual), and having that register set to 3 was the difference between my code working and not working.  The lack of any higher level documentation or functions related to this register is the main issue I am trying to raise.

o-marshmallow, 2025-04-25:
@jath03 Thanks for the clarification. I forwarded your message do the team responsible for the LL and HAL layers.
Actually, these APIs are not stable yet so they may change from one version to another.