espressif/esp-idf #15827: MCPWM Capture does not work as expected

AAJAY5, 2025-04-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# MCPWM Capture result are not expected

## Goal
- I am trying to measure 4-CH (PWM) input duty and frequency using ESP32. So, I have selected MCPWM capture to do this. 

## Input range
- Freq: 1Hz to 5000Hz
- Duty: 1% to 99%

## Other
- IDF: v5.4/release
- Board: ESP32

## Code

```c
#include <stdio.h>
#include <esp_log.h>
#include <esp_err.h>

/* FreeRTOS */
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

#define TAG "TEST"

#include "soc/mcpwm_periph.h"
#include "driver/mcpwm_prelude.h"
#include "driver/gpio.h"

typedef struct 
{
    int ch;
    int is_active;
    TimerHandle_t tmr;
    uint32_t high;
    uint32_t hhigh;
    uint32_t period;
    uint32_t tick;

    int next; 

    /* Calculate*/
    uint32_t freq;
    uint32_t duty;
}mcpwm_cap_data_t;

volatile uint32_t fail = 0;
static QueueHandle_t mcpwmQueue[4];
mcpwm_cap_data_t process_data[4];
mcpwm_capture_event_data_t edata;
uint32_t freq[4];
uint32_t duty[4];

#include "math.h"

static bool IRAM_ATTR mcpwm_cap(mcpwm_cap_channel_handle_t cap_channel, const mcpwm_capture_event_data_t *edata, void *user_data)
{
    QueueHandle_t queue = *(QueueHandle_t *)user_data;
    if (xQueueSendFromISR(queue, edata, 0) != pdPASS)
        fail++;
[... 220 lines cut ...]
I (20261) TEST: Fail 0
I (20261) TEST: 5000 5000 5000 5000, 40 40 10 40
I (21261) TEST: Fail 0
I (21261) TEST: 5000 5000 5000 5000, 40 40 10 40
I (22261) TEST: Fail 0
I (22261) TEST: 5000 5000 5000 5000, 40 40 10 40
I (23261) TEST: Fail 2
I (23261) TEST: 5000 5000 5000 5000, 40 40 10 40
I (24261) TEST: Fail 0
I (24261) TEST: 5000 5000 5000 5000, 40 40 10 40
I (25261) TEST: Fail 1
I (25261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (26261) TEST: Fail 0
I (26261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (27261) TEST: Fail 0
I (27261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (28261) TEST: Fail 0
I (28261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (29261) TEST: Fail 0
I (29261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (30261) TEST: Fail 1
I (30261) TEST: 5000 5000 12506 5000, 40 40 25 40
I (31261) TEST: Fail 0
I (31261) TEST: 5000 5000 12506 5000, 40 40 25 40
```

suda-morris, 2025-04-21:
Does it help if you reduce the frequency of the input signal? Maybe 5KHz input signal is close to the limit. 5KHz meaning to enter the interrupt every **100**us (because you enabled the pos and neg edge detection).

AAJAY5, 2025-04-21:
What is limit?

AAJAY5, 2025-04-22:
Thanks, I found issue. I have used 2 ESP32. One is used as generator to feed PWM output to capture ESP32. Basically both are connected to same PC.  So, I didn't common GND of both ESP32 as source is same. So at high frequency >2Khz random channel misses pulses. But when I GND common it will work as expected. I also removed `xQueueSendFromISR()` it takes some time that misses lower duty cycles.

closed 2025-04-22 by AAJAY5