espressif/esp-idf #15828: build-docs fails

truck-iguana, 2025-04-21:
I have been trying to create some visualizations to better understand the ecosystem. Previously, I was trying to create `call flow graph`'s using RTL files exported as `.expand` files, but I ran into issues with creating a good visualization via `Graphviz`. Perhaps that is a topic for another post.

For now, I've decided to check for any `doxygen` support. I am referencing [the documenting-code section of the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/documenting-code.html)

I have run the following commands:

```
$IDF_PATH/install.sh --enable-docs
cd $IDF_PATH/docs
build-docs -t esp32 -l en build
```

After 10-20 minutes, the builder finishes, and it returns in error.

```
en/esp32: (Check files sphinx-known-warnings.txt and $IDF_PATH/docs/_build/en/esp32/sphinx-warning-log.txt for full details.)
[2]

The following language/target combinations failed to build:
language: en, target: esp32, errcode: 2
```

## esp-idf version
```
idf.py --version
ESP-IDF v5.4.1
```

Checking the two files:

## sphinx-warning-log.txt
```
$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:30: WARNING: Invalid C++ declaration: Expected identifier in nested name. [error at 0]
  
  ^
$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:30: WARNING: Error in declarator or parameters-and-qualifiers
If pointer to member declarator:
  Invalid C++ declaration: Expected identifier in nested name. [error at 78]
    union esp_ble_mesh_prov_cb_param_t::ble_mesh_provisioner_prov_output_evt_param
    ------------------------------------------------------------------------------^
If declarator-id:
  Invalid C++ declaration: Expected identifier in nested name. [error at 78]
    union esp_ble_mesh_prov_cb_param_t::ble_mesh_provisioner_prov_output_evt_param
    ------------------------------------------------------------------------------^

$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:55: WARNING: Duplicate C++ declaration, also defined at api-reference/bluetooth/esp-ble-mesh:55.
Declaration is '.. cpp:member:: uint16_t model_id'.
$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:55: WARNING: Error in declarator or parameters-and-qualifiers
If pointer to member declarator:
  Invalid C++ declaration: Expected identifier in nested name. [error at 24]
    union esp_ble_mesh_model
    ------------------------^
If declarator-id:
  Invalid C++ declaration: Expected identifier in nested name. [error at 24]
    union esp_ble_mesh_model
    ------------------------^

$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:70: WARNING: Error in declarator or parameters-and-qualifiers
If pointer to member declarator:
  Invalid C++ declaration: Expected identifier in nested name. [error at 34]
    union esp_ble_mesh_device_delete_t
    ----------------------------------^
If declarator-id:
  Invalid C++ declaration: Expected identifier in nested name. [error at 34]
    union esp_ble_mesh_device_delete_t
    ----------------------------------^

$IDF_PATH/docs/_build/en/esp32/inc/esp_ble_mesh_defs.inc:73: WARNING: Error in declarator or parameters-and-qualifiers
If pointer to member declarator:
  Invalid C++ declaration: Expected identifier in nested name. [error at 35]
[... 188 lines cut ...]
esp_heap_trace.inc:line: WARNING: Error when parsing function declaration.
If the function has no return type:
  Invalid C++ declaration: Expected end of definition or ;. [error at 34]
    TAILQ_ENTRY (heap_trace_record_t) tailq
    ----------------------------------^
If the function has a return type:
  Error in declarator
  If declarator-id with parameters-and-qualifiers:
    Invalid C++ declaration: Expected identifier in nested name. [error at 12]
      TAILQ_ENTRY (heap_trace_record_t) tailq
      ------------^
  If parenthesis in noptr-declarator:
    Error in declarator or parameters-and-qualifiers
    If pointer to member declarator:
      Invalid C++ declaration: Expected '::' in pointer to member (function). [error at 32]
        TAILQ_ENTRY (heap_trace_record_t) tailq
        --------------------------------^
    If declarator-id:
      Invalid C++ declaration: Expecting "(" in parameters-and-qualifiers. [error at 32]
        TAILQ_ENTRY (heap_trace_record_t) tailq
        --------------------------------^

esp_heap_trace.inc:line: WARNING: Duplicate C++ declaration, also defined at api-reference/system/heap_debug:line.
Declaration is '.. cpp:type:: struct heap_trace_record_t heap_trace_record_t'.
```

o-marshmallow, 2025-04-22:
Hello @truck-iguana,

I just had a try on my end and was unable to reproduce the issue, despite using `v5.4.1` too. Make sure you don't have any local modification in your ESP-IDF directory and make sure to update the submodules before attempting to build the documentation:

```
git submodule update --recursive
```

ESP-Marius, 2025-04-22:
Different versions of doxygen sometimes gives different warnings, which might be why you are seeing them. Even then, it is just warnings, the built docs should still be there in the build folder.

I am not sure how building the docs relate to your goal though. What exactly are you trying to achieve? The pre-built docs can be accessed at https://docs.espressif.com/projects/esp-idf/en/stable/esp32/index.html

truck-iguana, 2025-04-22:
@o-marshmallow I tried again, `git submodule update --recursive` showed no output. And `git status` showed my repository was totally clean. I still got the same output

@ESP-Marius I see. The docs are there. The command itself returns in error, and if one checks the `$?` for a bash shell it shows the return code was `1` which indicates failure.

As for the pre-built docs, I understand. I have already been reading through them, then.

Although it seems like the build system may support my efforts, my goal is probably outside the scope of this project. I am trying to take an example program and build a call graph which I can use as a visual reference as I'm looking through documentation and actual source code.

## RTL files and call graph visualization

`idf.py menuconfig` -> Compiler options -> Dump RTL Files during compilation
This generates a plethora of `.expand` RTL files in the `build` directory.
Next, use a `.dot` file generator, such as [egypt](https://www.gson.org/egypt/egypt.html), and pass all of these files into it.
Then use a `.dot` file processor, such as [Graphviz](https://graphviz.org/) to generate a call graph.
I can show some examples below, but the visualization is not great and it takes a good amount of time to produce.

The first image is an overview, the second one is a close-up of some highly called lower layer functions (malloc and free, to no surprise). It would be a great resource in parallel with using example projects to gain deeper understanding of how the system works. Perhaps somebody else has worked on something similar and already successfully accomplished this?

[image]

[image]