espressif/esp-idf #15830: How can I improve the speed fwite to tfcard?

yizi15, 2025-04-22:
### Is your feature request related to a problem?

My project may write 18Mbit/s to tfcard. but it`s only 4Mbit/s in fact.

yizi15, 2025-04-23:
Now I not use the FatFS function f_write and write data by 'sdmmc_write_sectors', the speed abort 80Mbits/s. Why the f_write so slow?

igrr, 2025-04-23:
What is the size of the data block you are passing to f_write? What is the allocation unit size (cluster size) of the FAT filesystem? The larger these two values, the higher the writing speed you can typically get.

igrr, 2025-04-23:
Is the pointer you are passing to f_write aligned? Is it allocated in DMA-capable memory? What is the cluster size of FAT partition?

If f_write calls write_sectors multiple times, it may indicate a small cluster size (e.g. 4kB) or an almost full filesystem.

yizi15, 2025-04-23:
the 68000 pointer  aligned and  allocated in DMA-capable memory.  
```
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 1,
        .allocation_unit_size = 16 * 1024
    };
```
it's the format sd card argment, the sdmmc_write_sectors  call have some 16 blocks mult write(not alloc new memory), and other is 1block and may need alloc new buffer to sdio.

igrr, 2025-04-23:
Ok, given that the write size is 68000, there should be 4 full clusters written (16kB * 4), one partial cluster, two FAT updates, one directory chain update, and one FSInfo update.
If you log the size and pointer arguments to sdmmc_write_sectors, is this what you observe?

yizi15, 2025-04-23:
seem to right, but my log not save. However, the f_write is too slow (even only 1/20  raw write)

closed 2025-04-23 by yizi15