espressif/esp-idf #15831: ESP32-P4: 200 MHz PSRAM speed not in sdkconfig

Superberti, 2025-04-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should be possible to configure the PSRAM speed to 200 MHz.

### What is the actual behavior?

I'm evaluating the ESP32P4 Function_EV_Board (V1.4) and I started with the ESP-IDF 5.4.1/release and a brand new project (project wizard based on the hello_world example).
The problem I have is that I can't select anything else than 20 MHz PSRAM-speed, although 200 MHz should be possible. The "CONFIG_SPIRAM_SPEED_200M" flag doesn't exist in my sdkconfig and if I add it in the text editor it will be deleted next time I started menuconfig. The device target is set to esp32p4 and everything's compiling fine so far.
As I'd like to build a LVGL project with the 7 inch display I really need the 200 MHz option for a fast screen refresh.

Any ideas?

### Steps to reproduce.

1. Use ESP-IDF 5.4.1
2. Create a new project using ESP32P4
3. Try to configure the PSRAM speed to more than 20 MHz (20 MHz is the only option)

### Build or installation Logs.

```plain

```

peter-marcisovsky, 2025-04-22:
Hi @Superberti  you need to enable experimental features in the menucofnig first, to enable the 200 MHz PSRAM.

[image]

Superberti, 2025-04-22:
Wow, this was fast! 
Thank's for your help, now it's working like expected!