espressif/esp-idf #15833: BLE central sometimes drops sending of 4 consecutive connection interval packets at a time, causing disconnect when WiFi is active.

scholayil, 2025-04-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP32-C3 (QFN32) (version v0.4)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board with ESP32-C3-WROOM-02 module

### Power Supply used.

USB

### What is the expected behavior?

The mains USB powered esp32c3 device acts as a BLE central device. At this time it is connecting to only one coin cell powered BLE peripheral(Silabs BG2x), bonded.

We expect the connection to be stable and coin cell to last 2 years on the peripheral.

### What is the actual behavior?

1. We see peripheral initiated disconnects some times due to esp32c3 central not sending connection interval packets,
     * Within 2 hours to a day typically.
     * When this happens and we are able to log with Ellysis BT analyzer, we see the BLE central is not initiating empty connection interval packets up to 4 times within a supervisory period causing the peripheral to disconnect.

2. We have tried 16 seconds ble supervisory timeout and 4 second connection interval (call it 16s / 4s) 
i – 16s / 4s Disconnects often
ii. 10.5s / 1.5s - Disconnects less often
iii. 32s / 2s. – Has not disconnected in 3 days, but impractical value due to potential reconnect delay.
iv. 4s / 1s – Disconnects very often.

3. With  WiFi not connected( or provisioned) to AP, then BLE disconnects never happen if WiFi is init'd but not used. Even worst case above , iv. ( 4s / 1s) setting, has stable BLE connection without active wifi use.

### Steps to reproduce.

1. Set up one BLE central device and one bonded BLE peripheral connected to it.
2. Let the WiFi connect to an AP. We are also connected to aws mqtt server.
3. Try different pairs of supervisory and connection interval settings mentioned above (iv. is best setting to get dropped connection)
4. Monitor peripheral initiated connection drops with LED and or BT analyzer.

### Debug Logs.

```plain

```

### Diagnostic report archive.

We see 4 times as the max no of times the BLE central does not send connection interval packets to the peripheral. We are assuming it is because of WiFi priority.

Is there any other reason this happens?

Is it possible to adjust WiFi priority ?

esp-cjh, 2025-04-30:
Hi, @scholayil
Based on your description, it is confirmed that this is a Wi-Fi & BLE coexistence issue. Our colleagues responsible for coexistence will help investigate the root cause. Please provide the air packet capture file and complete log, along with the detailed Wi-Fi parameters and behavior. This will help us reproduce the issue.

Additionally, under Wi-Fi and BLE coexistence, with the parameters you provided, the BLE connection only has 4 transmission opportunities. We do not recommend using such settings. It is advised to set the timeout to at least 10 times the interval.

We apologize in advance that our response might be slightly delayed due to the May Day holiday.

esp-lrh, 2025-04-30:
For ESP32C3, when the BLE connection interval is large enough (typically, larger than 50 ms), BLE will always use high priority to request RF resource, and this priority can only preempted by WIFI Beacon RX. The reason why BLE disconnect often when the connection parameters are  16s / 4s or 4s / 1s may be due to conflicts with wifi beacons.

The default beacon interval of WIFI is 102.4 ms, therefore, it is recommended to set the connection interval to a value that is not a multiple of 102.4, such as `(n +0.3)*102.4` ms, `n` is an integer.

In addition, I will also provide a version specially optimized for this situation, so that BLE uses the highest priority when it is about to be disconnected. Please wait for a while.

esp-lrh, 2025-05-06:
Hi, @scholayil 

Please apply the patch and replace the BLE controller lib in `components/bt/controller/lib_esp32c3_family/esp32c3`
[patch.zip](https://github.com/user-attachments/files/20054571/patch.zip)

scholayil, 2025-05-09:
Thank you @esp-lrh , starting to test this solution. Will update.

scholayil, 2025-05-14:
@esp-lrh I am able to run with this patch and I have not had a disconnect with 4s connection interval, 16s supervisory interval and 0 latency. I have not had disconnects in about 6 hours with this. 

Before the changes I got 16 disconnects in 6 hours with the same settings. I will run it for a few days in real world situation and update.

To use this change in production what do we do? Will idf update with this change? Will it affect BT SIG filing QDIDs?

Thanks

esp-lrh, 2025-05-20:
@scholayil 

This change does not affect Bluetooth certification. If your product uses the same IDF version here (v5.4), you can apply the patch directly, if not, you can provide other IDF version and I wii provide the corresponding lib temporarily.

Since this change violates the original coexistence priority agreement, we still need to discuss whether to merge it into IDF as a formal change. It will not be updated to IDF so quickly for the time being.

esp-lrh, 2025-06-03:
@scholayil 

Oh, my mistake, the function I modified was not placed into IRAM, please use this new patch:
[patch_20250603.zip](https://github.com/user-attachments/files/20561949/patch_20250603.zip)

scholayil, 2025-06-09:
@esp-lrh Thank you. I will start testing this now.

scholayil, 2025-06-17:
@esp-lrh in 3.5 days, i got only 3 reconnects, two of them 4.7 minutes apart after 48 hours of run.  Not sure what the cause is yet.

scholayil, 2025-08-13:
Hi,

I am able to build the project with patch on my desktop. However, when I try to build on a build server, I get many linker errors. Full log in attached file. Any suggestions? thanks.

@esp-lrh 

/home/engr1/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /home/engr1/esp_5.4.0/esp-idf/components/bt/controller/esp32c3/bt.c:1841:(.text.esp_bt_controller_deinit+0x32): undefined reference to `chanSel_stack_enableSetCsaVsCmd'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

[patch-server-build-log.txt](https://github.com/user-attachments/files/21760069/patch-server-build-log.txt)

scholayil, 2025-08-14:
@esp-lrh what could be the difference between building on the desktop and on a build server with 5.4.0 ? Wont this happen even if I update to newer idf?

esp-lrh, 2025-08-15:
@scholayil 

I am not saying that you need to update the idf version, but this compilation problem should be due to the mismatch between the idf version and the lib version. Please check whether the idf version used by the build server is v5.4.0