espressif/esp-idf #15836: RMT: please enhance the documentation for signal_range_min_ns

fanfanlatulipe26, 2025-04-23:
RMT: please enhance the documentation for signal_range_min_ns

The message `(1113) rmt: rmt_receive(415): signal_range_max_ns too big`  is very frustrating because it is impossible to find a good information.   

But thanks to "grep, finstr, Google", deep in the software hierarchy of the ESP package and in the WEB ,
I discovered some information that **SHOULD** be added in the documentation.  

Please, add in the documentation in the description of rmt_rx_channel_config_t and rmt_receive_config_t some lines such as

> 
> Values for rmt_rx_channel_config_t::resolution_hz, rmt_receive_config_t::signal_range_min_ns, rmt_receive_config_t::signal_range_max_ns MUST satisfy some conditions.   
> The expression ( resolution_hz * signal_range_max_ns) / 1 000 000 000UL must be lower or equal to 65535 for ESP32,S2 and 32767 for S3, C3, C6 and H2.

For signal_range_min_ns I am not sure of the right message because I don't identify exactly   
rx_chan->filter_clock_resolution_hz   in the expression    
uint32_t filter_reg_value = ((uint64_t)rx_chan->filter_clock_resolution_hz * config->signal_range_min_ns) / 1000000000UL;

Nevertheless _signal_range_min_ns_ might be small and the user will nor fall in error...

suda-morris, 2025-04-23:
Thanks for the report. Indeed, driver should tell the max value that user can set when they see the error message.

fanfanlatulipe26, 2025-04-23:
Why not. Any way it will be pretty cool to have the information at hand.

janav1, 2025-04-24:
It would be great to do the same for **rmt_rx_channel_config_t.resolution_hz** because the maximum divider limit applies here too. It would be ideal to warn using the preprocessor #warning and not at runtime.

ginkgm, 2025-04-24:
Hi @fanfanlatulipe26 ,

The IDFGH-xxx in the title links to our issue tracker. Please don't touch it, or the synchronize between our internal system and Github updates may stop.

fanfanlatulipe26, 2025-04-24:
@ginkgm : got it ;-)

closed 2025-05-12 by espressif-bot