espressif/esp-idf #15839: ESP32-C6 - Wi-Fi-6 - Beamforming - How to enable the beamforming feedback?

shutterlore, 2025-04-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

### How do I enable WIFI beamformee functionality of ESP32-C6-WROOM-1?

Hi all, I'm new to this ESP32 development board and I use IDF version v5.5-dev-3226-g186f2a8929. 

I notice on the datasheet that it says this ESP32-C6-WROOM-1 chip supports IEEE 802.11ax downlink MU-MIMO and beamformee. I would like to use these features for my project.

I started with verifying the MIMO capabilities by running the wifi example "geting_started/station''. My AP is a R&S tester running on WLAN signaling mode, set to 802.11ax AP with 2x2 MIMO. The AP is beamformer-calable.

I built the example project as it is and successfully connected to my tester AP. However, when I examine the frames captured over the air, I notice the ESP32 does not advertise any of the HT/HE capabilities in its probe request frame. Additionally, when I proceed with the IEEE 802.11ax standard beamforming sounding procedure by sending out the NDP sounding packet to the ESP32, I do not observe the beamforming feedback frame.

In my compilation I did not change the menuconfig - component settings, neither did I find any settings related to beamforming.

My question is:

Can ESP32-C6-WROOM-1 respond to NDP sounding frames as beamformee? If so, how should I enable this feature?

Thank you in advance!

shutterlore, 2025-04-23:
Update on Apr 23.

In this fcuntion https://github.com/espressif/esp-idf/blob/465b159cd8771ffab6be70c7675ecf6705b62649/examples/wifi/getting_started/station/main/station_example_main.c#L120, I added few more entries:
```
.he_su_beamformee_disabled = 0,                        /**< Whether to disable support for operation as an SU beamformee. */
.he_trig_su_bmforming_feedback_disabled = 0,           /**< Whether to disable support the transmission of SU feedback in an HE TB sounding sequence. */
.he_trig_mu_bmforming_partial_feedback_disabled = 1,   /**< Whether to disable support the transmission of partial-bandwidth MU feedback in an HE TB sounding sequence. */
.he_trig_cqi_feedback_disabled = 1,  
```
according to the definition in https://github.com/espressif/esp-idf/blob/465b159cd8771ffab6be70c7675ecf6705b62649/components/esp_wifi/include/esp_wifi_types_generic.h#L580

However, this does not solve my problem. The probe request still lists no HE capabilities and no HE beamforming feedback frames were observed,

QingzhaoYin, 2025-04-24:
@shutterlore 
On the ESP32-C6, the HE Capabilities element is not included in the Probe Request frame, but it is present in the Association Request frame.

[image]

The ESP32-C6-WROOM-1 supports responding to NDP sounding frames as a beamformee by default.

If you do not observe the beamforming feedback frame, please share your packet capture (PCAP) file so we can help verify what's going on.

(Note: I'm not sure whether the ESP32 you mentioned earlier is the ESP32-C6. Regular ESP32 chips do not support Wi-Fi 6, and therefore do not support beamforming as a beamformee.)

shutterlore, 2025-04-24:
Hi, thanks for the response. 

> (Note: I'm not sure whether the ESP32 you mentioned earlier is the ESP32-C6. Regular ESP32 chips do not support Wi-Fi 6, and therefore do not support beamforming as a beamformee.)

Yes I am talking about the specific ESP32-C6-WROOM-1 chip in this entire issue.

> If you do not observe the beamforming feedback frame, please share your packet capture (PCAP) file so we can help verify what's going on.

The PCAP file is here [ESP32_with_testerAXAP.pcap.zip](https://github.com/user-attachments/files/19892019/ESP32_with_testerAXAP.pcap.zip)

Indeed I find those capabilities properly showed in the association request frames. 

[image]

However, the feedback is still not observed. [image]

The 74-byte long WLAN packet following the NDPA should be the NDP. There is no ESP32 response after this. I tried multiple sounding format, including non-TB-based/TB-based SU/CQI, neither of them worked. The sounding NDPA+NDP frames are generated from a R&S CMW270 tester; we have verified that they are correct by sounding 1) a laptop with Intel AX210 Wi-Fi adapter and 2) an Apple iPhone 13. (PCAP not available for these) 

[image]
[image]

So I think the sounding frames shouldn't be the problem at the first glance.

QingzhaoYin, 2025-04-25:
@shutterlore
In your packet capture, I noticed that the tester still sends NDPA frames to the ESP32-C6 while it is in sleep mode. However, since the ESP32-C6 is asleep, it cannot receive the NDPA.

I tested this by connecting the ESP32-C6 to a regular Wi-Fi router(ROG Rapture GT-AX11000). When the ESP32-C6 wakes up and sends a QoS NULL frame (with the power management bit cleared), the router only then starts sending NDPA frames. As a result, the ESP32-C6 is able to respond with feedback.

[image]

I recommend disabling power save mode on the ESP32-C6 during testing with the CMW270 to ensure it can properly receive NDPA frames and respond accordingly. Related API 'esp_err_t esp_wifi_set_ps(wifi_ps_type_t type)'

shutterlore, 2025-04-25:
@QingzhaoYin 
Thanks for the advice. The issue is indeed caused by the power saving mode. After I disable the power saving mode, my non-TB sounding frames were correctly responded by the ESP32-C6. I tried non-TB SU and non-TB CQI, they all worked. For TB sounding, however, no feedback frames were seen. Since I have clearly seen in the HE capabilities that TB sounding is supported, I wonder if there is other separate settings for TB as well?

The new capture is available here: [ESP32_with_testerAXAP_success.pcap.zip](https://github.com/user-attachments/files/19905462/ESP32_with_testerAXAP_success.pcap.zip) Sorry for the slightly messy capture since this one contains both the successful non-TB soundings and unsuccessful TB soundings. You may use the wireshark filter `wlan.fc.type_subtype == 0x0012` to explicitly locate every sounding trigger that is used for TB sounding.

The sounding sequences are fired manually from the tester in single-shots so the intervals will be different. The following picture shows what I set for the sounding properties for the TB SU format on the tester. Non-TB works well with the same GI and LTF numbers. I set Nc = 1 because the ESP32-C6 is 1T1R so only 1 spatial stream supported. 
[image]

QingzhaoYin, 2025-04-25:
@shutterlore Could you please check if other devices (e.g., the Intel AX210 Wi-Fi adapter and the Apple iPhone 13) respond to TB sounding? It’s possible that the TB sounding feedback frames are missing in the capture because the AID was not properly configured.

shutterlore, 2025-04-25:
@QingzhaoYin I can only verify them next monday. The AID of my ESP32-C6 should have no problem because it's always 1 and the tester had only 1 association back then. I revisited the TB NDPA in my capture and noticed that they always broadcast two station info, where one is AID 1 and the other is AID 438. This looks very stange since I was sending TB SU sounding frames so I expect only 1 station being nominated. The subsequent trigger BRP correctly targets AID=1 though.
[image]

I would imagine this to be an abnormal behaviour, but since it's from the tester, I can't really change it. Nevertheless, getting non-TB SU to work is good enough for me, TB will be a bonus but not a must.

shutterlore, 2025-04-28:
I confirm that the TB NDP sounding (SU/CQI) does not work on the Intel AX210 equipped laptop. The same strange AID issue is observed.

Then the TB sounding issue would most like be the problem of the tester itself rather than ESP32. I would use an SDR to play some pre-generated Wi-Fi samples containing NDPA+NDP+SIFS+BFRP later on to further verify the TB functionality. 

Nontheless, thank you again for your help.

closed 2025-04-30 by shutterlore