espressif/esp-idf #15840: esp_mesh_send doesn't fully respect esp_mesh_send_block_time

dvirassulin, 2025-04-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-PICO-V3-02 (revision v3.0)

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32

### Power Supply used.

External 3.3V

### What is the expected behavior?

When calling `esp_mesh_send()`,  the method should be blocking for the time configured using the method `esp_mesh_send_block_time()`. e.g., If called `esp_mesh_send_block_time(5000)`, `esp_mesh_send()` should not take longerthan 5 seconds to return.

### What is the actual behavior?

In our project, we are measuring the time it takes `esp_mesh_send()` to execute, and sometimes even though we called `esp_mesh_send_block_time(5000)` it takes more than 30 seconds to return.

### Steps to reproduce.

This bug is hard to reproduce, since it seems circumstantial. Attaching a simplified code so the settings are clear:
1. Step 1 - configure and initialize:
```
        static const uint8_t MESH_ID[6] = { 0x77, 0x77, 0x77, 0x77, 0x77, 0x77};

        /* tcpip initialization */
        ESP_ERROR_CHECK(esp_netif_init());
        /* create network interfaces for mesh (only station instance saved for further manipulation, soft AP instance ignored */
        ESP_ERROR_CHECK(esp_netif_create_default_wifi_mesh_netifs(&_stationNetInterface, nullptr));
        /* wifi initialization */
        wifi_init_config_t config = WIFI_INIT_CONFIG_DEFAULT();
        ESP_ERROR_CHECK(esp_wifi_init(&config));
        ESP_ERROR_CHECK(esp_wifi_set_storage(WIFI_STORAGE_FLASH));
        if (esp_netif_set_hostname(_stationNetInterface, "myHostName") != ESP_OK){
            SG_WARN("Failed setting hostname");
        }
        ESP_ERROR_CHECK(esp_wifi_start());
        ESP_ERROR_CHECK(esp_wifi_set_ps(wifi_ps_type_t::WIFI_PS_NONE));
        ESP_ERROR_CHECK(esp_wifi_get_mac(wifi_interface_t::WIFI_IF_STA, &_stationMacAddress.mac[0]));
        /* mesh initialization */
        ESP_ERROR_CHECK(esp_mesh_init());

        /* set mesh topology */
        ESP_ERROR_CHECK(esp_mesh_set_topology(esp_mesh_topology_t::MESH_TOPO_TREE));
        /* set network capacity */
        ESP_ERROR_CHECK(esp_mesh_set_capacity_num(100));
        /* set mesh max layer according to the topology */
        ESP_ERROR_CHECK(esp_mesh_set_max_layer(25));
        /* set vote percentage threshold for approval of being a root */
        ESP_ERROR_CHECK(esp_mesh_set_vote_percentage(0.9));
        /* set the size of the packets receiving queue */
        ESP_ERROR_CHECK(esp_mesh_set_xon_qsize(32));
        /* set mesh softAP associate expired time */
        ESP_ERROR_CHECK(esp_mesh_set_ap_assoc_expire(30));
        /* Disable mesh power-saving function -
           TODO - if relevant, need to add supporting configuration */
        ESP_ERROR_CHECK(esp_mesh_disable_ps());

#pragma GCC diagnostic push
#pragma GCC diagnostic ignored "-Wmissing-field-initializers"
        mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();
#pragma GCC diagnostic pop

        /* mesh ID */
        memcpy((uint8_t *) &cfg.mesh_id, MESH_ID, 6);
        /* router */
        cfg.allow_channel_switch = true;
        cfg.router.ssid_len = static_cast<std::uint8_t>(strcpy_safe(cfg.router.ssid, "mySSID"));
        strcpy_safe(cfg.router.password, "myPassword");
        /* mesh softAP */
        ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(wifi_auth_mode_t::WIFI_AUTH_WPA2_PSK));
        cfg.mesh_ap.max_connection = 3;
        cfg.mesh_ap.nonmesh_max_connection = 0;
        strcpy_safe(cfg.mesh_ap.password, "ApPassword");
        ESP_ERROR_CHECK(esp_mesh_set_config(&cfg));
        ESP_ERROR_CHECK(esp_mesh_allow_root_conflicts(false)); // Avoid multiple roots
```
2. step 2: set blocking time
        `ESP_ERROR_CHECK(esp_mesh_send_block_time(5000));`
3. step 3: start mesh
        `ESP_ERROR_CHECK(esp_mesh_start());`
4. Step4: send a message using `esp_mesh_send` and time it:
In a separate task and in a loop, we send some payloads according to the application's demands, and we time and log it:
```
        auto upTimeBefore = std::chrono::duration_cast<TimeSpan>(std::chrono::microseconds(esp_timer_get_time()));
        esp_mesh_send(/*some destination*/, /*some data*/, MESH_DATA_P2P, NULL, 0);
        auto upTimeAfter = std::chrono::duration_cast<TimeSpan>(std::chrono::microseconds(esp_timer_get_time()));
        auto timeToexecute = upTimeAfter - upTimeBefore;
        ESP_LOGI("esp_mesh_send took %d milliseconds to execute", timeToexecute.count());
```

then we see the time it took to execute is way beyond 5 seconds that were configured. we saw it take up to 30 seconds.

### Debug Logs.

```plain

```

### More Information.

In our project there are multiple esp32 devices communicating with each other and with external IPs. this issue happens in some of them, maybe due to network interferences but we don't know for sure. All the devices have the same cinfigurations.

dvirassulin, 2025-05-07:
@zhangyanjiaoesp I don't think so (That is, we have our own logs that are not esp's, and the relevant parts are reported here).
but maybe we can try and get them in the next time it happens. What kind of logs do you need? can you give me a log configuration that will give you that?

zhangyanjiaoesp, 2025-05-08:
@dvirassulin 
What I need is the info-level log printed during normal operation. The mesh configuration you mentioned earlier looks fine, and since you also said the issue seems to be environment-related, I’d like to see if the log can provide any clues.