espressif/esp-idf #15841: ESP32: BT fails to re-enable BB after stop/start

rojer, 2025-04-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-KIT

### Power Supply used.

USB

### What is the expected behavior?

BLE should work after stopped and restarted (without reboot)

### What is the actual behavior?

BLE starts but doesn't work properly: no errors are reported by advertisements are not send and scan returns nothing

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain

```

rojer, 2025-05-07:
ping @esp-zhp @weiyuhannnn

esp-zhp, 2025-05-08:
BLE is divided into the controller and the host. Based on your description:

> “Start the host (e.g., NimBLE), enable advertising
> Check that advertisements are sent with a BLE scanner app
> Stop the host (without rebooting the device)
> Start the host again, enable advertisements
> Device is no longer visible”

It seems you are restarting the **host**.
However, based on your later description, it sounds like you are actually restarting the **controller**.
Could you clarify this part more clearly?

By the way, in practical use,
when shutting down the controller, you should first call:
`esp_bt_controller_disable`
followed by
`esp_bt_controller_deinit`

And when reusing it later,
you should first call:
`esp_bt_controller_init`
and then
`esp_bt_controller_enable`

Also, if possible, could you share your code? That would help make the issue much clearer.

rojer, 2025-05-08:
correct, we stop the host with `ble_hs_stop()` and when that's done, we call `nimble_port_stop()` and `nimble_port_deinit()`.
`nimble_port_deinit()` internally calls `esp_bt_controller_disable()` and `esp_bt_controller_deinit()`, so it all checks out.

we then start the controller again with `nimble_port_init()` and `ble_hs_start()`.

it works fine on C3 and C6 and on ESP32 it doesn't because of what i described above.

unfortunately, it's not easy for me to share the code i use, as it's not only based on IDF and uses our own internal framework, i'd need to construct a special IDF-only app.

esp-zhp, 2025-05-14:
You can create a dedicated demo for us to reproduce the issue — this will greatly help us in identifying and resolving the problem.

rojer, 2025-09-17:
ping @esp-zhp

esp-lrh, 2025-09-24:
Hi, @rojer 

When coexisting with WiFi, the issue of not re-initializing BTBB does exist. In early versions, disabling BT would not reset BTBB, so the issue did not appear. After this change [ad3554f](https://github.com/espressif/esp-idf/commit/ad3554fbed715c9cc64fe6db2a81b51770f63504#diff-5b74f40b3ddfdab435166a3e2e99ddc8324560abd7eb6cb6c6a7dfa6535145e1R103), BTBB will be reset when controller init/deinit. The fixed that add `s_time_phy_rf_just_enabled = 0;` in `esp_bt_controller_enable` is appropriate, and I will fix this issue in IDF later. 

Thank you for your feedback, If you have any other questions, feel free to ask