espressif/esp-idf #15846: Thread Border Router does not support DNS query forwarding to upstream resolver

tom-a-wagner, 2025-04-24:
### Is your feature request related to a problem?

We are currently evaluating using ESP32s for implementing a Thread Border Router, and are currently using the ESP Thread Border Router DevKit and the [esp-thread-br](https://github.com/espressif/esp-thread-br) example code.

The border router itself and devices in the Thread network are unable to resolve "normal" (non-mDNS/DNS-SD) DNS queries using the default DNS resolver, which is the border router.
The border router replies to these queries with "Not Found", even if the resolved hostname exists on the internet.

The issue is that that the border router does not act as a recursive DNS resolver.
The OpenThread configuration [OPENTHREAD_CONFIG_DNS_UPSTREAM_QUERY_ENABLE](https://openthread.io/reference/config/group/config-dnssd-server#openthread_config_dns_upstream_query_enable) is not enabled, which would make the border router forward queries for unknown hostnames to its upstream DNS resolver.
Enabling this option requires the implementation of the OpenThread [DNS Platform Abstraction functions](https://openthread.io/reference/group/plat-dns), which ESP-IDF currently doesn't implement.

The Thread specification (at least v1.4) specifies that a Thread border router MUST implement a recursive DNS resolver, so currently an esp-idf border router would not conform to that part of the specification unless the functions are implemented and the option enabled manually.

### Describe the solution you'd like.

ESP-IDF implements the DNS platform abstraction functions.
Existing platform abstractions seem to be implemented at https://github.com/espressif/esp-idf/tree/master/components/openthread/src/port.

ESP-IDF then adds a configuration option to turn on `OPENTHREAD_CONFIG_DNS_UPSTREAM_QUERY_ENABLE` to enable this.
To conform to the thread specification by default, this option should likely be turned on automatically for border routers, so when the esp-idf config option `OPENTHREAD_BORDER_ROUTER` is turned on.

### Describe alternatives you've considered.

As a workaround, devices in the thread network could be set up to also try another non-esp-idf border router (if available), or public resolvers, e.g. Google DNS or Cloudflare DNS, if resolving via the ESP border router fails.
This requires that all devices in the network can have their software modified.

chshu, 2025-04-24:
@tom-a-wagner The feature has been supported since commit: https://github.com/espressif/esp-idf/commit/2c01990a6279f2aab7676678f97d81a45b05c6d4, and `OPENTHREAD_CONFIG_DNS_UPSTREAM_QUERY_ENABLE` is enabled for BR by default, could you please fetch the latest code and test the feature again.

tom-a-wagner, 2025-04-24:
Thanks for the quick response, it does indeed work with the latest code.

Thanks!