espressif/esp-idf #15856: `ets_ecdsa_verify` Undefined in Application Build with ESP32-C6 Rev.0 chip on a custom board and Secure Boot V2

diegomartinezvdi, 2025-04-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When building an application targeting **ESP32-C6 Rev.0** with **ESP-IDF v5.4** and **Secure Boot V2** enabled, the build fails with an **undefined reference to `ets_ecdsa_verify()`**, even though `CONFIG_ESP_ROM_REV0_HAS_NO_ECDSA_INTERFACE=y` is set.

The fallback implementation in `bootloader_ecdsa.c` is only compiled under `BOOTLOADER_BUILD` or `CONFIG_APP_BUILD_TYPE_RAM`, and is therefore missing in regular app builds — even when required.

---

## Environment

- **Target**: ESP32-C6 Rev.0
- **ESP-IDF Version**: v5.4
- **Toolchain**: riscv32-esp-elf-gcc (14.2.0)
- **Secure Boot V2**: Enabled
- **Flash Encryption**: Enabled (Using ECDSA)

---

## Problem

When running `idf.py build`, the linker fails: `undefined reference to ets_ecdsa_verify esp-idf/bootloader_support/libbootloader_support.a(secure_boot_signatures_app.c.obj): in function calculate_image_public_key_digests: secure_boot_signatures_app.c: undefined reference to ets_ecdsa_verify
`

This occurs because:

- `secure_boot_signatures_app.c` is compiled and linked into the **application**.
- It uses `ets_ecdsa_verify()` to validate public key digests.
- For ESP32-C6 Rev.0, this symbol **does not exist in ROM**.
- The fallback implementation exists in `bootloader_ecdsa.c`, but that file is:
  - Only compiled under `BOOTLOADER_BUILD` or `CONFIG_APP_BUILD_TYPE_RAM`.
  - Not included in normal app builds.
- So the app linker fails to find `ets_ecdsa_verify()`.

---

## Expected Behavior

When `CONFIG_ESP_ROM_REV0_HAS_NO_ECDSA_INTERFACE=y`, and app-level code references `ets_ecdsa_verify()`, the build system should:

- Automatically compile and link `bootloader_ecdsa.c` into the application, **even if not building the bootloader**.

---

## Attempts to resolve the issue that have failed.

1. We tried building the bootloader before the app using `idf.py all`, as well as separate steps like `idf.py bootloader && idf.py app` and `idf.py bootloader && idf.py build`.
2. Updated to ESP-IDF v5.4.1, and it still fails with the same error.

mahavirj, 2025-04-25:
@diegomartinezvdi We will fix this issue, please use following patch interim:

```patch
diff --git components/bootloader_support/CMakeLists.txt components/bootloader_support/CMakeLists.txt
index d6520afd1d0..bae865dc3b4 100644
--- components/bootloader_support/CMakeLists.txt
+++ components/bootloader_support/CMakeLists.txt
@@ -65,6 +65,9 @@ if(CONFIG_APP_BUILD_TYPE_APP_2NDBOOT)
 endif()
 
 list(APPEND srcs "src/bootloader_sha.c")
+if(CONFIG_ESP_ROM_REV0_HAS_NO_ECDSA_INTERFACE)
+    list(APPEND srcs "src/${IDF_TARGET}/bootloader_ecdsa.c")
+endif()
 
 if(BOOTLOADER_BUILD OR CONFIG_APP_BUILD_TYPE_RAM)
     set(include_dirs "include" "bootloader_flash/include"
@@ -82,10 +85,6 @@ if(BOOTLOADER_BUILD OR CONFIG_APP_BUILD_TYPE_RAM)
         list(APPEND srcs "src/bootloader_utility_tee.c")
     endif()
     list(APPEND priv_requires hal)
-    if(CONFIG_ESP_ROM_REV0_HAS_NO_ECDSA_INTERFACE)
-        list(APPEND srcs
-            "src/${IDF_TARGET}/bootloader_ecdsa.c")
-    endif()
 else()
     set(include_dirs "include" "bootloader_flash/include")
     set(priv_include_dirs "private_include")
```

diegomartinezvdi, 2025-04-29:
Thank you, @mahavirj , for your assistance. The patch worked perfectly. I appreciate your help!

closed 2025-04-29 by diegomartinezvdi

matteofumagalli1275, 2025-05-25:
Will this fix be included in which SDK?