espressif/esp-idf #15861: RMT Conflict on ESP32-S3 with Different Resolutions

ElishaBentzi, 2025-04-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

54.3.20+develop.sha.d81abd0

### Espressif SoC revision.

ESP32S3 240MHz, 320KB RAM, 16MB Flash 

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

LILYGO_T7_S3

### Power Supply used.

USB

### What is the expected behavior?

same CLOCK, RMT Different TX channels can have different resolutions

### What is the actual behavior?

The Higher resolution is used for all the Channels.

Problem Description:
using RMT for two different purposes simultaneously:
Synchronized PWM generation with two high-resolution TX channels (e.g., 80 MHz).
OneWire (DS18B20) communication using the RMT-based onewire_bus library (which uses 1 MHz resolution and TX/RX channels).
both parts of the code use automatic RMT channel assignment (because the .channel_id field is not available in the RMT configuration structures in my version of the Arduino framework).
Describe the symptom: When RMT PWM (80 MHz) initialization is enabled, the OneWire library fails during device detection (specifically during onewire_bus_reset) with errors such as 1-wire reset pulse receive timeout and/or rmt_receive(...): channel not in enable state.
If i disable RMT PWM initialization, OneWire communication works correctly.

### Steps to reproduce.

#include <Arduino.h>
#include <driver/rmt_tx.h> // Principal header for TX
#include <driver/gpio.h>

// --- Depuración Estilo Personalizado ---
#define DEBUG_HAAS true
#if DEBUG_HAAS
  #define DEBUG_SERIAL_PRINT(...) Serial.printf(__VA_ARGS__)
#else
  #define DEBUG_SERIAL_PRINT(...) do {} while (0)
#endif

// --- Configuración del Ejemplo ---
#define EXAMPLE_RMT_TX_GPIO_0   GPIO_NUM_2
#define EXAMPLE_RMT_TX_GPIO_1   GPIO_NUM_11
#define EXAMPLE_RMT_RESOLUTION_HZ 10 * 1000 * 1000 // 10 MHz
#define EXAMPLE_RMT_TRANS_QUEUE_SIZE 4
#define EXAMPLE_RMT_MEM_BLOCK_SYMBOLS 64
[... 305 lines cut ...]

    if (burst_err != ESP_OK) {
      DEBUG_SERIAL_PRINT("[LOOP] Failed to start PWM burst.\r\n");
      // Esperar un poco antes de reintentar en caso de error
      vTaskDelay(pdMS_TO_TICKS(1000));
    } else {
       vTaskDelay(pdMS_TO_TICKS(10));
    }
  } else {
    // RMT está ocupado, ceder CPU
    vTaskDelay(pdMS_TO_TICKS(1000));
  }

  
}

### Debug Logs.

```plain
Initializing OneWire bus... OneWire RMT bus created successfully. 
[DS18B20] Device iterator created, start searching... 
E (1511) 1-wire.rmt: onewire_bus_rmt_reset(393): 1-wire reset pulse receive timeout 
E (1511) 1-wire.device: onewire_device_iter_get_next(61): reset bus failed 
E (1514) rmt: rmt_receive(419): channel not in enable state
```

ElishaBentzi, 2025-04-26:
ESP32-S3 (QFN56) (revision v0.1)

Kainarx, 2025-04-27:
Hi @ElishaBentzi ! Thank you for the report.
You can refer the [TRM](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/rmt.html) for channel resolution
[image]
And I am not sure whether `54.3.20+develop.sha.d81abd0` is a valid IDF version.
You can rebase to master and try switching the order of the application channels. 
Alternatively, can the MCPWM generate the PWM waveform you need?

suda-morris, 2025-04-27:
Hi @ElishaBentzi Please check if your IDF version contains this enhancement: https://github.com/espressif/esp-idf/commit/555bfd8f5c9e3c440886a563c7f3933b8eba63e5

ElishaBentzi, 2025-04-28:
Sorry i really dont know how to check for commits, i try latest available pioarduino V5.4.1 and developer channel.  the same problem, i believe that the resolution difference from 80MHz in one channel to 1MHz in another channel is too problematic to support by the prescaler. then i decide to use the suggestion and move my code to use MCPWM.

suda-morris, 2025-04-29:
I constructed the following use case, one TX channel with resolution 80MHz, and another RX channel with resolution 1MHz (same as your application).

```c
    rmt_tx_channel_config_t tx_channel_cfg = {
        .mem_block_symbols = SOC_RMT_MEM_WORDS_PER_CHANNEL,
        .gpio_num = TEST_RMT_GPIO_NUM_A,
        .clk_src = RMT_CLK_SRC_DEFAULT,
        .resolution_hz = 80 * 1000 * 1000,
        .trans_queue_depth = 1,
    };
    rmt_channel_handle_t tx_channel = NULL;
    rmt_rx_channel_config_t rx_channel_cfg = {
        .mem_block_symbols = SOC_RMT_MEM_WORDS_PER_CHANNEL,
        .gpio_num = TEST_RMT_GPIO_NUM_B,
        .clk_src = RMT_CLK_SRC_DEFAULT,
        .resolution_hz = 1 * 1000 * 1000, // 1MHz
    };
    rmt_channel_handle_t rx_channel = NULL;
    rmt_new_tx_channel(&tx_channel_cfg, &tx_channel);
   rmt_new_rx_channel(&rx_channel_cfg, &rx_channel);
```

The code runs without problem with enhancement made by commit https://github.com/espressif/esp-idf/commit/555bfd8f5c9e3c440886a563c7f3933b8eba63e5

The enhancement has also been backported to release/v5.4 (https://github.com/espressif/esp-idf/commit/ff8c836be3a60fcb660ef420d67069d7f08701fd) and release/v5.3 (https://github.com/espressif/esp-idf/commit/e98ded4e7beec29a6423ec6ac647221f44927c44) already. All you need is to switch to the latest release branch (or just use the master branch).

closed 2025-04-29 by suda-morris