espressif/esp-idf #15863: Can’t initialize GPIO 27 which is USB D+

Jayliu0712, 2025-04-27:
ESP32H2
ESP-IDF 5.3.2

When I initialize USB + (GPIO27 pin), an error occurs.

> --- Error: GetOverlappedResult failed (PermissionError(13, '鎷掔粷璁块棶銆?', None, 5))

What causes this problem?

Bruce297, 2025-04-27:
Hi @Jayliu0712  This error usually means that the serial port is being used by another program (like a serial terminal or in the Arduino IDE). 
Double-check that you’ve closed all the programs that might be establishing a serial communication with your ESP board. Then, unplug and plug back your ESP board and try again.

Jayliu0712, 2025-04-27:
> Hi [@Jayliu0712](https://github.com/Jayliu0712) This error usually means that the serial port is being used by another program (like a serial terminal or in the Arduino IDE). Double-check that [...]

@Bruce297 I tried according to your suggestions, but the problem still persists. I intend to use the GPIO27 pin as an ordinary I/O port for other purposes. Here is part of my schematic diagram.

[test.pdf](https://github.com/user-attachments/files/19928849/test.pdf)

Bruce297, 2025-04-27:
On ESP32H2, GPIO 26 and 27 are used by USB-Serial-JTAG by default. In order to use them as GPIOs, USB-Serial-JTAG will be disabled by the drivers.

Jayliu0712, 2025-04-27:
> On ESP32H2, GPIO 26 and 27 are used by USB-Serial-JTAG by default. In order to use them as GPIOs, USB-Serial-JTAG will be disabled by the drivers.

@Bruce297 Strangely, in the question at the following link, I was able to control the USB - related I/O. Are there any points to note in the design of the hardware peripheral circuits of the ESP32H2?

https://github.com/espressif/esp-idf/issues/15797

Jayliu0712, 2025-04-28:
> On ESP32H2, GPIO 26 and 27 are used by USB-Serial-JTAG by default. In order to use them as GPIOs, USB-Serial-JTAG will be disabled by the drivers.

@Bruce297 Sorry，I think this is an issue with the hardware peripheral circuit because I got the same problem when I re - wired to other I/O ports. I'm sorry to trouble you. You can close this issue.

closed 2025-04-28 by Jayliu0712