espressif/esp-idf #15872: Can't Set SPI Clock Frequency more than 20 MHz on ESP32-P4

NaitikSanas, 2025-04-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-P4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-P4 Function EV Board

### Power Supply used.

USB

### What is the expected behavior?

Hello, So I ported my ST7789 Display driver to ESP32-P4 and it appears that SPI Clock is fixed to 20 MHz only. 

### What is the actual behavior?

I can see display render the UI at 20MHz SPI Clock but its literal slideshow. Setting any value above 20MHz always fails to initialize SPI Instance: 

### Steps to reproduce.

This is initialization i used in my driver :
```c
spi_device_interface_config_t devcfg;
	memset(&devcfg, 0, sizeof(devcfg));
	//devcfg.clock_speed_hz = SPI_Frequency;
	devcfg.clock_speed_hz = SPI_MASTER_FREQ_80M;
	devcfg.queue_size = 7;
	//devcfg.mode = 2;
	devcfg.mode = 3;
	devcfg.flags = SPI_DEVICE_NO_DUMMY;

	if ( GPIO_CS >= 0 ) {
		devcfg.spics_io_num = GPIO_CS;
	} else {
		devcfg.spics_io_num = -1;
	}
	
```

### Debug Logs.

```plain
E (723) spi_master: spi_bus_add_device(426): invalid sclk speed
actual freq - 21
```

### More Information.

I tried spi master example as well and ran into same issue. Now I did went through documentation saw [this section ](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32p4/api-reference/peripherals/spi_master.html#gpio-matrix-and-io-mux) too, but unfortunately pin 9 is not available on header so i can't confirm if issue is due to not setting native spi pins. I could not find information if there are other native SPI compatible pins available on header so i am creating this issue. Hope i get any resolution here. 

Thanks.

wanckl, 2025-04-29:
@NaitikSanas  `devcfg.clock_source = SPI_CLK_SRC_SPLL` 👍

NaitikSanas, 2025-04-29:
@wanckl Hi That actually resolves the issue! Thank you very much :) I transitioned from 4.4.7 so didn't knew now you also need to set clock source every time.

closed 2025-04-29 by NaitikSanas