espressif/esp-idf #15873: How to set the transmission power and receiving sensitivity of ESP32H2?

aaa147147, 2025-04-29:
### Is your feature request related to a problem?

I found from sdkconfig and the latest ESP-IDF SDK that the ESP32H2 has not yet set the transmission power and receiving sensitivity
[image]

### My needs are as follows:
**1. How to correctly set the transmission power of ESP32H2, base on openthread
2. Can the receiving sensitivity be adjusted, and where can it be adjusted specifically?**

Thanks！

aaa147147, 2025-05-12:
Can someone help me? T-T

chshu, 2025-05-15:
You can change the transmission power by calling [esp_ieee802154_set_txpower](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/include/esp_ieee802154.h#L78).

The receiving sensitivity is not configurable. If you do want to filter the lower RSSI frame, you can read the RSSI value from the received frame, then compare and drop, please refer to [esp_ieee802154_receive_done()](https://github.com/espressif/esp-idf/blob/master/components/ieee802154/include/esp_ieee802154.h#L529).

Alvin1Zhang, 2025-07-02:
Thanks for reporting, will close due to inactivity, feel free to reopen.

closed 2025-07-02 by Alvin1Zhang