espressif/esp-idf #15876: ESP32S2 ADC line fitting calibration scaling is wrong when using DB 12 attenuation

kevinhikaruevans, 2025-04-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32S2

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

When using DB12 attenuation, I would expect a raw ADC value of 4095 to give about 2500 mV.

### What is the actual behavior?

It seems to give about 1/2 the expected value. In this case, 1280 mV.

### Steps to reproduce.

Compiling for the ESP32S2, you can run this sample code

```c
void test() {
  // adc_cali_check_scheme returns ADC_CALI_SCHEME_VER_LINE_FITTING on esp32s2

  adc_cali_line_fitting_config_t cali_config = {
      .unit_id = ADC_UNIT_1,
      .atten = ADC_ATTEN_DB_12,
      .bitwidth = ADC_BITWIDTH_12,
  };
  adc_cali_handle_t handle = NULL;

  ESP_ERROR_CHECK(adc_cali_create_scheme_line_fitting(&cali_config, &handle));

  int voltage = 0;
  ESP_ERROR_CHECK(adc_cali_raw_to_voltage(handle, 4095, &voltage)); 

  ESP_LOGI(TAG, "read voltage=%d", voltage); // returns 1280 mV
}
```

Here, 4095 is the max ADC value (12-bit) when running under continuous mode.

### Debug Logs.

```plain
In console:

read voltage=1280
```

### More Information.

I am using platformio here FYI. I'm using ESP IDF 5.3.1 but the equivalent is framework-espidf@3.50301.0. Full PIO config:

```ini
[env]
platform_packages =
    platformio/tool-openocd-esp32@2.1200.20230419
    platformio/toolchain-xtensa-esp-elf@13.2.0+20240530
    platformio/framework-espidf@3.50301.0
    platformio/tool-esptool@1.413.0
    platformio/tool-esptoolpy@1.40501.0

monitor_raw = yes
monitor_speed = 115200

# Add extra bits
extra_scripts =
build_flags = -Wall -Wno-nonnull-compare # -DCOMPONENT_EMBED_FILES=shared/...

# Debug
debug_tool = esp-prog
debug_speed = 20000
debug_init_break = tbreak app_main
debug_init_cmds = ftdi_tdo_sample_edge falling

# Upload
upload_protocol = esp-prog
upload_speed = 20000

board = board_rev1
platform = espressif32@6.8.1
framework = espidf
```

Related thread: https://esp32.com/viewtopic.php?t=45406

kevinhikaruevans, 2025-04-29:
Tried it with 5.4.1 (platformio/framework-espidf@3.50401.0) and applying this patch https://github.com/espressif/esp-idf/commit/b64ddb18939d06426607a04816e6de881524e87f and it still results in the same issue

Bruce297, 2025-05-05:
Hi @kevinhikaruevans 
ESP32S2 ADC onshot only supports 13 bits, cali_data=1280 is correct when raw_data=4095, if you want a max value, raw data should be 8191.

Also,  even on other chips, only set `adc_cali_line_fitting_config_t.bitwidth` is invalid, you need to set `adc_oneshot_chan_cfg_t.bitwidth` too. 

Also be careful, S2 DB12 effective measurement range is 0 ~ 2300.

kevinhikaruevans, 2025-05-05:
Hi @Bruce297 

I'm currently using the ADC's continuous mode which is limited to 12 bits. For context, I'm initializing the continuous config like:

```c
  adc_continuous_config_t cont_config = {
      .pattern_num = 1,
      .adc_pattern =
          (adc_digi_pattern_config_t[1]){
              (adc_digi_pattern_config_t){
                  .unit = ADC_UNIT_1,
                  .channel = ADC_CHANNEL_0,
                  .atten = ADC_ATTEN_DB_12,
                  .bit_width = ADC_BITWIDTH_12,

              },
          },
      .sample_freq_hz = 1000,
      .conv_mode = ADC_CONV_SINGLE_UNIT_1,
      .format = ADC_DIGI_OUTPUT_FORMAT_TYPE1,
  };
```

Then when I call `adc_continuous_read` and read the type1 data, it returns 4095 when the ADC is saturated. Here's the code to how I'm initializing it https://esp32.com/viewtopic.php?t=45406

kevinhikaruevans, 2025-07-24:
Just wanted to see if there were any updates on this. 

FYI using `ADC_BITWIDTH_13` results in `E (509) adc_continuous: adc_continuous_config(448): ADC bitwidth not supported`