espressif/esp-idf #15877: httpd_uri.c functions aren't thread safe

gonzzor, 2025-04-29:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

Registering / unregister handlers from multiple tasks isn't thread safe, nor is it to unregister handlers on a running webserver.

Reading the documentation for the functions hints that this should be possible. It doesn't forbid it from what I can tell.

If the webserver is running and one unregisters a handler, there is a risk of crash as well if the memory is free() while httpd_uri() is trying to use such entry.

### What is the actual behavior?

Random crashes in the httpd_uri.c files when registering handlers from multiple tasks running on different cores.

For example here https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/components/esp_http_server/src/httpd_uri.c#L150 the code does malloc() and now the uri struct member points to invalid memory. If some other task now tries to see if a handler is already registered it will try to access this random memory.

Similar issue can be found all over this file.

### Steps to reproduce.

1. Start two tasks, preferably on different cores.
2. Have both task register uri handlers
3. Get timing correct and watch things crash, likely on a strlen or memcpy.

### Debug Logs.

```plain

```

### More Information.

I've currently worked around this issue locally by having an external mutex around these calls to avoid multiple tasks to register/unregister URIs at the same time.

hrushikesh430, 2025-05-20:
Hi @gonzzor, I tried to reproduce to your issue but on my side it is working. I will attach the patch file with this comment. I made the changes in http_server/simple example to start the two server as task and unregistered the URI handler and tried to invoke the URI. Everything works fine.

[multiple_server_tasks.patch](https://github.com/user-attachments/files/20332722/multiple_server_tasks.patch)

May be you can send your patch, so that will help me to reproduce and debug the code easily.

gonzzor, 2025-05-21:
Hi @hrushikesh430,

Thanks for looking into this. Seems that I wasn't detailed enough in the report.

I'm only running one httpd server in total. However, I have two (or three) tasks that call httpd_register_uri_handler() which causes crashes for me.

Perhaps I've misunderstood the API and that it's not thread safe to begin with? The documentation only mentions "_URI handlers can be registered in real time as long as the server handle is valid._"

Regardless, this piece of code isn't thread safe between an external caller and the httpd server thread.
https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/components/esp_http_server/src/httpd_uri.c#L148-L180

As the malloc() is directly inserted into the `hd->hd_calls[]` array, the httpd server could be receiving a request at this time and iterate through the array, considering the entry to be valid, i.e. `!= NULL`. The data in it is still some random that could cause crashes.

hrushikesh430, 2025-06-02:
To keep the HTTP server simple and reduce memory overhead, thread safety has not been implemented. Applications are responsible for managing thread safety at their own layer. We will update the documentation to reflect this.

gonzzor, 2025-06-02:
> To keep the HTTP server simple and reduce memory overhead, thread safety has not been implemented. Applications are responsible for managing thread safety at their own layer. We will update the documentation to reflect this.

Thanks for confirming this.

closed 2025-06-24 by espressif-bot