espressif/esp-idf #15879: RISC V ULP Coprocessor shared variable linker error

0xPiyush, 2025-04-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1 (release version)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Project builds successfully.

### What is the actual behavior?

The project fails to build with linker errors when linking ulp_main.elf. VS Code on windows, project open in docker dev container through vscode, clean build ran inside docker dev container using the build button in vscode also tested using idf.py build (same result).

### Steps to reproduce.

Minimal reproducible example:

main.c
```c
#include <stdio.h>

#include "esp_log.h"
#include "esp_sleep.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "ulp_main.h"
#include "ulp_riscv.h"

// ONLY the shared variables needed by the simplified ULP code
RTC_DATA_ATTR uint16_t active_buf;
// RTC_DATA_ATTR uint16_t idx0;
// Keep types simple for testing

// ULP binary symbols
extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[] asm("_binary_ulp_main_bin_end");

static const char* TAG = "MAIN_APP";

static void init_ulp_program(void) {
    esp_err_t err = ulp_riscv_load_binary(
        ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
    ESP_ERROR_CHECK(err);
    err = ulp_riscv_run();
    ESP_ERROR_CHECK(err);
    ESP_LOGI(TAG, "Done initializing ULP-RISC-V program!");
}

void app_main(void) {
    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();

    if (cause != ESP_SLEEP_WAKEUP_ULP) {
        ESP_LOGI(TAG, "Normal boot");
[... 275 lines cut ...]
[8/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_gpio.c.obj
[9/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_adc.c.obj
[10/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_uart.c.obj
[11/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_touch.c.obj
[12/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_interrupt.c.obj
[13/17] Building C object CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_i2c.c.obj
[14/17] Linking C executable ulp_main.elf
FAILED: ulp_main.elf 
: && /opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -Os -march=rv32imc_zicsr_zifencei -mdiv -fdata-sections -ffunction-sections -march=rv32imc_zicsr_zifencei --specs=nano.specs --specs=nosys.specs   -nostartfiles -Wl,--gc-sections -Wl,--no-warn-rwx-segments -Wl,-Map=/workspaces/test-project/build/esp-idf/main/ulp_main/ulp_main.map -T /workspaces/test-project/build/esp-idf/main/ulp_main/ulp_riscv.ld /opt/esp/idf/components/ulp/ld/esp32s3.peripherals.ld CMakeFiles/ulp_main.dir/workspaces/test-project/main/ulp/main.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_vectors.S.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/start.S.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_adc.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_lock.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_uart.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_print.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_i2c.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_utils.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_touch.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_gpio.c.obj CMakeFiles/ulp_main.dir/opt/esp/idf/components/ulp/ulp_riscv/ulp_core/ulp_riscv_interrupt.c.obj -o ulp_main.elf   && :
/opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: CMakeFiles/ulp_main.dir/workspaces/test-project/main/ulp/main.c.obj: in function `main':
main.c:(.text.startup.main+0x0): undefined reference to `active_buf'
/opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: main.c:(.text.startup.main+0x4): undefined reference to `active_buf'
/opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: main.c:(.text.startup.main+0x12): undefined reference to `active_buf'
/opt/esp/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: main.c:(.text.startup.main+0x30): undefined reference to `ulp_set_wakeup_period'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
[837/1076] Linking C static library esp-idf/wpa_supplicant/libwpa_supplicant.a
ninja: build stopped: subcommand failed.

 *  The terminal process "ninja" terminated with exit code: 1.
```

### Diagnostic report archive.

[idf-diag-a460c0a8-9cb8-4f11-9d0e-a1b507c83e9d.zip](https://github.com/user-attachments/files/19970124/idf-diag-a460c0a8-9cb8-4f11-9d0e-a1b507c83e9d.zip)

sudeep-mohanty, 2025-04-30:
HI @0xPiyush,
You need not define the global variable as `extern` in the ULP RISC-V app if the intention is to have it as a shared variable between the main core and the ULP core. You can see an example [here](https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/examples/system/ulp/ulp_riscv/gpio/main/ulp/main.c#L27) and read about it in the API reference [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/ulp-risc-v.html#accessing-the-ulp-risc-v-program-variables).

0xPiyush, 2025-04-30:
@sudeep-mohanty Thank you for the quick response, normally when you declare a variable in the RISC V ULP code it becomes available to the main core via the generated header file and all the variables are accessible with the ulp_ prefix that raises a few questions for me.

1. Since these variables are created in the main core using RTC_DATA_ATTR will they be available to both the cores without the ulp_ prefix?
2. Does the main core still need to include the generated header file?

Finally, what about ulp_set_wakeup_period() this function also fails to link when I use it in the risc v ulp core, even after including ulp_common.h it still fails to link this function. Is this not available to the ULP core? If it's not available how can I schedule wakeup for ULP from within the ULP? Any work arounds or other ways to do it without the help of main core?

Again thank you for your time.

sudeep-mohanty, 2025-04-30:
@0xPiyush 

> Since these variables are created in the main core using RTC_DATA_ATTR will they be available to both the cores without the ulp_ prefix?

No, this is not the case. Variables declared on the main core, irrespective of the attribute, are not available to the ULP core because it does not have access to the linker file used by the main core. The ULP core does not know where in the RTC memory is the variable placed. The opposite is possible because the ULP project is compiled first and the [esp32ulp_mapgen.py](https://github.com/espressif/esp-idf/blob/master/components/ulp/esp32ulp_mapgen.py) does special processing during the build stage to make global symbols on the ULP app available to main core.

> Does the main core still need to include the generated header file?

Yes, we always need to include the generated header file since all global ULP symbols are declared here.

> Finally, what about ulp_set_wakeup_period() this function also fails to link when I use it in the risc v ulp core, even after including ulp_common.h it still fails to link this function. Is this not [...]

You are right that `ulp_set_wakeup_period()` fails to link on the ULP core because it is only available to the main core. You would need to setup the wakeup period once from the main core and do not need to set the wakeup period again from within the ULP. Once the ULP goes into halt (i.e., returns from the `main()`), the ULP timer begins the countdown until the next wakeup period and wakes up the ULP subsequently.

0xPiyush, 2025-04-30:
Got it, Thank you for your quick responses.

closed 2025-04-30 by 0xPiyush