espressif/esp-idf #15883: Facing issue flash encryption+secure boot sbv2 combination in development mode for esp32s3

keshamodhia, 2025-04-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,
I am working for the Flash Encryption and Secure boot v2, where i am facing the issue

**Module - ESP32-S3
IDF - 5.0.7
Building Hello World Example given in idf
Platform - windows, using cmd [command prompt] commands for building binaries**

Work Done so Far,
**Enabled Secure boot** in menuconfig **[devlopment mode]**
**Generated signed key** for the images
**Tickmarked sign image during build time in menuconfig**
Upgraded the partitioon table as per requirement
**Flashed bootloader.bin manually and flashed other application.bin, partition.bin with usin idf.py flash**
Secure boot is done successgfully

As I am not able to read the contents from the flash and also unsigned images are not supported bu the module
Tested Only signed Bootloade.bin and upon that signed bianries only ups the board and allows to run the default hello world example

Now Trying with the **Flash encryption**

**Enabled flash encryption**
as want to enabled both at the same time
and again dumped the **bootloader.bin** in the module **[this time not dumped the application.bin after duming the bootloader.bin]**

Attaching the logs and sdkconfig

[20250425_VM_flash_encryp_logs_SINGLE_FINAL_Reset_logs.txt](https://github.com/user-attachments/files/19974727/20250425_VM_flash_encryp_logs_SINGLE_FINAL_Reset_logs.txt)

[sdkconfig.txt](https://github.com/user-attachments/files/19974776/sdkconfig.txt)

Kindly reply on high priority

AdityaHPatwardhan, 2025-05-26:
Hi @keshamodhia, Apologies for the delayed reply.
After secure boot is enabled, the ability to read disable an eFuse block is restricted and hence Flash Encryption cannot be enabled after that.
Please see the third point in [Restrictions after enabling secure boot](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/secure-boot-v2.html#restrictions-after-secure-boot-is-enabled)

AdityaHPatwardhan, 2025-06-03:
@keshamodhia, I am closing the issue since the above response should resolve it.
Please feel free to re-open if required.
Thanks

closed 2025-06-03 by AdityaHPatwardhan