espressif/esp-idf #15885: esp32-s3 接usb 5G wifi模组

bachekaiqilai, 2025-04-30:
### Is your feature request related to a problem?

C5客户接受不了价格，希望s3能使用5G频段，请问esp32-s3能接usb wifi模组吗？像Linux那些插上usb后做网卡联网

MaxwellAlan, 2025-04-30:
Hi @bachekaiqilai 

S3 可以通过 esp-hosted 方案外接 C5 实现 5G WiFi, S3 本身作为 Host MCU 使用即可。

目前我们尚未支持过其他厂商的 WiFi 网卡，当然，不排除有其他厂商的 WIFI 网卡可能可以支持 S3 作为 Host MCU。

bachekaiqilai, 2025-04-30:
> 你好[@bachekaiqilai](https://github.com/bachekaiqilai)
> 
> S3 可以通过 esp-hosted 方案外接 C5 实现 5G WiFi，S3 本身作为 Host MCU 使用即可。
> 
> 目前我们尚未支持其他厂商的 WiFi 中断，当然，不排除其他厂商的 WIFI 中断可能可以支持 S3 作为主机 MCU。

您好，请问要移植WIFI 网卡能在S3跑，需要在S3上面大概做什么开发呢？WiFi网卡驱动移植到S3与S3硬件能支持的可能性有多大呢？依照您的预估2个人把整个流程弄完的开发周期会大于半年以上吗？期望大佬回复，您的回复对选型和后续开发有极大极大超级大帮助

bachekaiqilai, 2025-05-07:
> 从您目前所提供的信息来看，还比较难以评估。 如果方便的话，可以提供一下可能需要使用的 5G WiFi 网卡信息： 1.不同的硬件通信接口（SPI, USB, UART 等）移植的难度相差很大； 2.不同的协议栈架构划分，开放的难度完全不同，例如 S3 上是否需要运行部分协议栈（LwIP, mac80211 等），或者完全不需要网络协议栈等，类似使用 AT 固件等外挂无线网卡 3.通信的性能，延迟是否有要求等？ PS：有个疑问，为什么一定要外挂 5G 无线网卡，另外 S3+外挂 5G网卡的成本 真的会比 C5 单芯片的方案 更优惠吗？

[HF-LPD110 Dual Band Wi-Fi Module User Manual EN-V1.1(20230818)(1).pdf](https://github.com/user-attachments/files/20080771/HF-LPD110.Dual.Band.Wi-Fi.Module.User.Manual.EN-V1.1.20230818.1.pdf)
客户要跑（语音模型 或者是音频流+websocket） + 5G网，C5 单芯片除了5G，别的做不到。s3+c5成本客户又接受不了....
上传了一个at固件的相关信息pdf，您看能否评估这样能否满足实现。感谢大哥

MaxwellAlan, 2025-05-08:
@bachekaiqilai 
看起来 S3+LPD110 通过 UART AT 命令看起来能实现 5G 的联网功能，但是对于 音频流和websocket 通信流量这块，不确定 UART 的方式能否满足 Throughput 需求，这个你们需要进一步评估看下。

bachekaiqilai, 2025-05-26:
> [@bachekaiqilai](https://github.com/bachekaiqilai) 看起来 S3+LPD110 通过 UART AT 命令看起来能实现 5G 的联网功能，但是对于 音频流和websocket 通信流量这块，不确定 UART 的方式能否满足 Throughput 需求，这个你们需要进一步评估看下。

大哥您好，我这边已经打板测试通过使用串口通过at指令连接了5GWiFi，要跑一个小智这样的ai语音对话，应该怎么把芯片的2.4G wifi 联网变成使用at的5G模组联网呢？

MaxwellAlan, 2025-08-07:
> > [@bachekaiqilai](https://github.com/bachekaiqilai) 看起来 S3+LPD110 通过 UART AT 命令看起来能实现 5G 的联网功能，但是对于 音频流和websocket 通信流量这块，不确定 UART 的方式能否满足 Throughput 需求，这个你们需要进一步评估看下。
> 
> I'm really interested in what the issue said. I hope the authorities can come up with some interesting and doable guidance on how to put this function into practice.

The approach in this issue is not a conventional practice, and using it this way cannot bring into play the advantages of the IDF network protocol stack. It is recommended that developers with 5G requirements can choose the ESP32-C5 chip; for those with AI computing needs, they can select ESP32-S3, or the esp-hosted solution of ESP32-P4 + ESP32-C5.

closed 2025-08-07 by MaxwellAlan