espressif/esp-idf #15886: fwrite dosesn't work with data allocated in PSRAM

aygh4266, 2025-04-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,

I am using PSRAM to allocate an array of nearly 1 MB where a FW Image can be stored. The FW Image file has been sent via BLE successfully.
`EXT_RAM_BSS_ATTR uint8_t FW_Data[FW_SIZE];`

When trying to write the FW Image in spiffs using the fwrite function it doesn't seem to work.
the fileSize will be received from client via BLE and stored in the variable uint32_t fileSize

Any help would be appreciated

code:
``` 
f = fopen(filepath, "wb");  // "wb" for write binary 
if(f ==nullptr)
{
    return ESP_FAIL;
}
size_t bytesWritten = fwrite(FW_Data, 1, fileSize, f);
ESP_LOGI(tagCHR, "LOG after fwrite: size of FW_Data %d, written bytes %d, file size %" PRIu32, sizeof(FW_Data), (int)bytesWritten, fileSize);
```

ESP-Debug-LOGS:
```
I (40333) BT-STM32Callback: LOG after fwrite: size of FW_Data 1048576, written bytes 0, file size 107108
```

aygh4266, 2025-04-30:
Assumption: SPI Flash and PSRAM use the same SPI bus and therefore cannot be addressed simultaneously.

igrr, 2025-04-30:
> When trying to write the FW Image in spiffs using the fwrite function it doesn't seem to work.

Could you please clarify what is happening instead? (File is written but doesn't contain data; file not written at all but fwrite returns a non-zero value; exception or panic happens; etc...)

Which chip and which ESP-IDF version are you using? Could you attach your sdkconfig file?

If you don't receive the data over BLE and fill the FW_Data array with some random values, does the issue also occur? (If yes, could you provide a project to reproduce the issue?

> SPI Flash and PSRAM use the same SPI bus and therefore cannot be addressed simultaneously.

spi_flash driver does have some special handling for source buffers located in PSRAM, so this use case is expected to work... however there is a chance that it doesn't under some specific conditions. Your answers to the questions above may help identify the cause.

aygh4266, 2025-04-30:
@igrr This is my sdkconfig file
[sdkconfig.txt](https://github.com/user-attachments/files/19978658/sdkconfig.txt)

AAJAY5, 2025-05-01:
```sketch
#include <stdio.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_check.h"
#include "esp_littlefs.h"
#include "esp_vfs_fat.h"
#include <unistd.h> // for unlink()
const char *base_path = "/fs";

static const char *TAG = "EXAMPLE";

static esp_err_t app_littlefs_init()
{
    ESP_LOGI(TAG, "Initializing LittleFS");

    esp_vfs_littlefs_conf_t conf = {
        .base_path = base_path,
        .partition_label = "storage",
        .format_if_mount_failed = true,
        .dont_mount = false,
    };

    // Use settings defined above to initialize and mount LittleFS filesystem.
    // Note: esp_vfs_littlefs_register is an all-in-one convenience function.
    esp_err_t ret = esp_vfs_littlefs_register(&conf);

    if (ret != ESP_OK)
    {
        if (ret == ESP_FAIL)
        {
            ESP_LOGE(TAG, "Failed to mount or format filesystem");
        }
        else if (ret == ESP_ERR_NOT_FOUND)
        {
            ESP_LOGE(TAG, "Failed to find LittleFS partition");
        }
        else
        {
            ESP_LOGE(TAG, "Failed to initialize LittleFS (%s)", esp_err_to_name(ret));
        }
        return ret;
    }

    size_t total = 0, used = 0;
    ret = esp_littlefs_info(conf.partition_label, &total, &used);
    if (ret != ESP_OK)
    {
        ESP_LOGE(TAG, "Failed to get LittleFS partition information (%s)", esp_err_to_name(ret));
    }
    else
    {
        ESP_LOGI(TAG, "Partition size: total: %d, used: %d", total, used);
    }

    return ret;
}

static esp_err_t app_fatfs_init()
{
    const esp_vfs_fat_mount_config_t mount_config = {
        .max_files = 4,
        .format_if_mount_failed = true,
        .allocation_unit_size = CONFIG_WL_SECTOR_SIZE,
        .use_one_fat = false,
    };
    ESP_LOGI(TAG, "Mounting FAT filesystem in read/write mode");

    static wl_handle_t wl_handle = WL_INVALID_HANDLE;
    esp_err_t err = esp_vfs_fat_spiflash_mount_rw_wl(base_path, "storage", &mount_config, &wl_handle);
[... 91 lines cut ...]
I (461) efuse_init: Min chip rev:     v0.0
I (464) efuse_init: Max chip rev:     v0.99 
I (468) efuse_init: Chip rev:         v0.2
I (472) heap_init: Initializing. RAM available for dynamic allocation:
I (479) heap_init: At 3FC96A20 len 00052CF0 (331 KiB): RAM
I (484) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (489) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (494) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (499) esp_psram: Adding pool of 1792K of PSRAM memory to heap allocator
I (506) spi_flash: detected chip: winbond
I (509) spi_flash: flash io: dio
W (512) spi_flash: Detected size(16384k) larger than the size in the binary image header(8192k). Using the size in the binary image header.
I (525) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (531) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (538) main_task: Started on CPU0
I (550) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (550) main_task: Calling app_main()
I (552) EXAMPLE: Successfully allocated 1MB buffer in PSRAM at 0x3c040898

I (559) EXAMPLE: Mounting FAT filesystem in read/write mode
I (634) EXAMPLE: File deleted successfully: /fs/test.bin

I (634) EXAMPLE: Creating file /fs/test.bin
I (18825) EXAMPLE: Written: 1048576, FileSize: 1048576
```

AAJAY5, 2025-05-01:
> ### Answers checklist.
> * [x]  I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.[x]  I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.[x]  I have searched the issue tracker for a similar issue and not found a similar issue.
> 
> ### General issue report
> Hello everyone,
> 
> I am using PSRAM to allocate an array of nearly 1 MB where a FW Image can be stored. The FW Image file has been sent via BLE successfully. `EXT_RAM_BSS_ATTR uint8_t FW_Data[FW_SIZE];`
> 
> When trying to write the FW Image in spiffs using the fwrite function it doesn't seem to work. the fileSize will be received from client via BLE and stored in the variable uint32_t fileSize
> 
> Any help would be appreciated
> 
> code:
> 
> ```
> FILE* f = fopen(filepath, "wb");  // "wb" for write binary 
> if(f ==nullptr)
> {
>     return ESP_FAIL;
> }
> size_t bytesWritten = fwrite(FW_Data, 1, fileSize, f);
> ESP_LOGI(tagCHR, "LOG after fwrite: size of FW_Data %d, written bytes %d, file size %" PRIu32, sizeof(FW_Data), (int)bytesWritten, fileSize);
> ```
> 
> ESP-Debug-LOGS:
> 
> ```
> I (40333) tagCHR: LOG after fwrite: size of FW_Data 1048576, written bytes 0, file size 107108
> ```

ESP_LOGI(tagCHR, "LOG after fwrite: size of FW_Data %d, written bytes %d, file size %" PRIu32, sizeof(FW_Data), (int)bytesWritten, fileSize);
I (40333) tagCHR: LOG after fwrite: size of FW_Data 1048576, written bytes 0, file size 107108
```c
// fileSize > sizeof(FW_Data). You are accessing invalid data/ranges
size_t bytesWritten = fwrite(FW_Data, 1, fileSize, f);
```

Fi

igrr, 2025-05-02:
> The file is being written and opened successfully but it doesn't seem to contain any data because fwrite returns zero.
> No exception or panic happens

One additional point to check, when fwrite returns zero, what is the value of `errno`?

aygh4266, 2025-05-02:
> ## I have perform test and it works for me. Taking global buffer in PSRAM give some compile error so I've used malloc in PSRAM.

Thank you very much. I will attach my full code next week. 

Background: I‘m receiving the FW data via BLE using NimBLE C++ and storing it in a global buffer, since the data is being chunked and written in chunks. With every new chunk I am incrementing the buffer offset. 

I’m using global buffer because I want to use the same data in another function to store it in SPIFFS. 

When I used a buffer allocated in internal RAM, it works. But I had to store every single chunk after receiving it via BLE and append it to the written data in spiffs due to my limited internal RAM size

aygh4266, 2025-05-02:
> One additional point to check, when fwrite returns zero, what is the value of `errno`?

I will check this next week and keep you updated. Thanks for your help. Have a nice weekend!

Please keep this ticket open

aygh4266, 2025-05-06:
@AAJAY5 Thank you very much. I have checked your code and it only worked once. After flashing the esp32s3 again, it didn't work. 
Have you tried it multiple times ??

AAJAY5, 2025-05-06:
> [@AAJAY5](https://github.com/AAJAY5) Thank you very much. I have checked your code and it only worked once. After flashing the esp32s3 again, it didn't work. Have you tried it multiple times ??

Log please.

aygh4266, 2025-05-06:
> > [@AAJAY5](https://github.com/AAJAY5) Thank you very much. I have checked your code and it only worked once. After flashing the esp32s3 again, it didn't work. Have you tried it multiple times ??
> 
> Log please.

I have integrated our code in my nimble callbacks. Here is my code and log 

```cpp
FILE* f;	
	sprintf(filepath, "/spiffs/%s", filename);	
	  
	if (strlen(filepath) > MAX_FILE_PATH)
	{
		ESP_LOGE(tagCHR,"File path too long");
		///@todo implement logic for too long path
		return ESP_FAIL; 
	}

	if (unlink(filepath) == 0) {
        ESP_LOGI(tagCHR, "File deleted successfully: %s\n", filepath);
    } else {
        ESP_LOGE(tagCHR, "Error deleting file");
    }

	f = fopen(filepath, "wb");  // "wb" for write binary 
	ESP_LOGI(tagCHR, "Filepath : %s", filepath);
	ESP_LOGI(tagCHR, "Opened file in wb only at first time");

	errno = 0;
	ESP_LOGI(tagCHR, "LOG Before fwrite: size of FW_Data %d, file size %" PRIu32, sizeof(stmDataVal), fileSize);
	size_t bytesWritten = fwrite(FW_Data , 1, fileSize, f);	//only write the file size and not size of FW_Data 
	ESP_LOGI(tagCHR, "LOG after fwrite: size of FW_Data %d, written bytes %d, file size %" PRIu32, sizeof(stmDataVal), (int)bytesWritten, fileSize);

	fclose(f);  // close the file after writing
	
	if (bytesWritten != fileSize) {
		ESP_LOGE(tagCHR, "fwrite failed, errno: %d -> %s", errno, strerror(errno));
		checkSpaceOnSPIFFS();
		ESP_LOGE(tagCHR, "fwrite wrote %d bytes, expected %" PRIu32, bytesWritten, fileSize);
		ESP_LOGE(tagCHR, "Failed to write entire file to SPIFFS");	
		flatBufferOffset = 0;
		return ESP_FAIL;
	}
	ESP_LOGI(tagCHR, "Successfully wrote %d bytes", (int)bytesWritten);
	return ESP_OK;	
}
```

```cpp
I (145580) tagCHR: Deleting all Files ........
I (145890) tagCHR: Deleted file: Blinky.bin
I (146000) tagCHR: Deleted file: test.bin
E (146140) tagCHR: Error deleting file
I (146190) tagCHR: Filepath : /spiffs/test.bin
I (146190) tagCHR: Opened file in wb only at first time
I (146190) tagCHR: LOG Before fwrite: size of FW_Data 4, file size 524288
I (146800) tagCHR: LOG after fwrite: size of FW_Data 4, written bytes 0, file size 524288
E (146800) tagCHR: fwrite failed, errno: 28 -> No space left on device
E (146800) tagCHR: fwrite wrote 0 bytes, expected 524288
E (146810) tagCHR: Failed to write entire file to SPIFFS
I (146830) tagCHR: SPIFFS total: 956561, used: 251, free: 956310
```
@igrr errno shows space issue but I have enough space on spiffs.

aygh4266, 2025-05-06:
@AAJAY5 So testing it multiple times shows that it works sometimes and sometimes not. Maybe it is a fragmanting issue ?

AAJAY5, 2025-05-06:
- Add the below code to function and also where you allocating FW_Data? I would recommended to allocated in `app_main` 
```c
esp_err_t storeInSPIFFS(NimBLECharacteristic* pCharacteristic, uint16_t connHandle)
{
      if(FW_Data == NULL){
          ESP_LOGE(tagCHR, "malloc failed");
          return ESP_FAIL;
     }
//.... your code

```

```c
// main.c app_main()

void *FW_Data = NULL;
void app_main  (){
FW_Data = heap_caps_malloc(FW_SIZE, MALLOC_CAP_SPIRAM);
if(FW_Data == NULL){
    ESP_LOGE(TAG, "malloc failed");
    }
}
```

aygh4266, 2025-05-15:
Still notworking with PSRAM, So I allocate the data firstly in internal RAM and used fwrite to store it in SPIFFS. The data will be chunked. The chunk will be stored temporarily in RAM

closed 2025-05-15 by aygh4266

AAJAY5, 2025-05-15:
Most probably there must be some memory issue like out-of-bound somewhere.