espressif/esp-idf #15887: Nimble pairing fails for some/most devices with  encryption change event BLE_HS_ENOTCONN and BLE_ERR_CONN_TERM_MIC

florianhumblot, 2025-04-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4.1-535-g986a52a52f

### Espressif SoC revision.

ESP32-C3 (rev 4)

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I expect all phones to be able to pair to Nimble when using Just Works security and reading encrypted characteristics.

### What is the actual behavior?

The connection is dropped when attempting to pair on most phones I have to test with.

### Steps to reproduce.

1. Connect to the ESP32
2. Read a encrypted characteristic through the nRF Connect app
3. Allow your phone to pair

### Debug Logs.

```plain

```

### More Information.

Attached are the application debug logs of nimble, I turned off all the other tasks so it's really only nimble. All of the phones did the same process:
 1. Open the nRF connect app
 2. Scan for devices and connect to the ESP32
 3. Attempt to read an encrypted characteristic
 4. Allow pairing when the system prompts you to.

The bonding process fails on android 15 with a pixel 9 and 9 pro:
[esp_nimble_bond_fail_log_android_15_pixel_9.txt](https://github.com/user-attachments/files/19979104/esp_nimble_bond_fail_log_android_15_pixel_9.txt)
It also fails with an iphone 14 pro (iOS 18.3.2):
[esp_nimble_bond_fail_log_ios_18.3.2_iphone_14_pro.txt](https://github.com/user-attachments/files/19979106/esp_nimble_bond_fail_log_ios_18.3.2_iphone_14_pro.txt)

But the pairing succeeds on samsung S24 Ultra!
[esp_nimble_bond_success_log_android_14_samsung_s24_ultra.txt](https://github.com/user-attachments/files/19979105/esp_nimble_bond_success_log_android_14_samsung_s24_ultra.txt)

[... 199 lines cut ...]
  memset(&advParams, 0, sizeof(advParams));
  advParams.conn_mode = BLE_GAP_CONN_MODE_UND;
  advParams.disc_mode = BLE_GAP_DISC_MODE_GEN;

  resultCode = ble_gap_adv_start(blehrAddrType, nullptr, BLE_HS_FOREVER, &advParams, GapEvent, nullptr);
  if (resultCode != 0) {
    ESP_LOGE(LOG_TAG, "error enabling advertisement; rc=%d, reason: %s\n", resultCode, NimbleErrToString(resultCode));
    return;
  }
}
```

In case you want to see a characteristic: 
```c++
{
    .uuid = &uuidBleVersion.u,
    .access_cb = AccessCallback,
    .arg = reinterpret_cast<void*>(static_cast<uintptr_t>(BleCharacteristics_e::BLE_VERSION)),
    .descriptors = nullptr,
    .flags = BLE_GATT_CHR_F_READ | BLE_GATT_CHR_F_READ_ENC,
    .min_key_size = 0,
    .val_handle = nullptr,
    .cpfd = nullptr,
},
```

rahult-github, 2025-05-01:
Hi @florianhumblot , 

Can you please try attached changes  

IDF patch to be applied at $IDF_PATH : 0001-fix-bt-Update-bt-lib-for-ESP32-C3-and-ESP32-S3-edf92.txt

unzip the contents of folder esp32c3.zip to components/bt/controller/lib_esp32c3_family/esp32c3. 

[0001-fix-bt-Update-bt-lib-for-ESP32-C3-and-ESP32-S3-edf92.txt](https://github.com/user-attachments/files/19999892/0001-fix-bt-Update-bt-lib-for-ESP32-C3-and-ESP32-S3-edf92.txt)

[esp32c3.zip](https://github.com/user-attachments/files/19999895/esp32c3.zip)

florianhumblot, 2025-05-02:
@rahult-github awesome, those patches fix the issues on both the iPhone and the Google Pixel :)

rahult-github, 2025-05-05:
Change has been merged to master and relevant backports too merged.

closed 2025-05-05 by rahult-github

esp-zhp, 2026-01-20:
Thanks for reporting this issue.

I have tested locally using the ESP-IDF version you provided and tried to reproduce the problem, but unfortunately I was not able to reproduce the MIC failure on my side.

could you please help test whether the problem can be reproduced using the following official example?

```
examples/esp-idf/examples/bluetooth/bluedroid/ble/gatt_security_server
```

This example uses a standard BLE Secure Connections bonding flow and minimal application logic. If the MIC failure (0x3D) can still be reproduced with this example on Google Pixel devices, it would be very helpful evidence for us to further investigate a potential issue in the ESP32-C3 Bluetooth stack or controller libraries.

Please let us know your test results. Thanks for your cooperation.

[image]

cpatulea, 2026-04-20:
For reference, the commit associated with this issue was https://github.com/espressif/esp-idf/commit/0df0b53b0a94828a2ed24fd198b4401ec7e311a6