espressif/esp-idf #15889: PEEK UART buffer

wilmer-suarez-simplisafe, 2025-04-30:
### Is your feature request related to a problem?

Is there a way to PEEK at the UART buffer without consuming the data?

`uart_read_bytes` seems to be the only way to do read the buffer; but I would like to just peek at the data within the buffer so that I can intercept AT commands for a functionality that I'm working on for specific commands.

suda-morris, 2025-05-08:
@wilmer-suarez-simplisafe 
Maybe you will get inspired by how "peek" is handled in the vfs driver:

https://github.com/espressif/esp-idf/blob/aaebc374676621980878789c49d239232ea714c5/components/esp_driver_uart/src/uart_vfs.c#L73

nopnop2002, 2025-05-10:
The same effect can be achieved by transferring the data read by uart_read_char to other tasks using a Queue or MessageBuffer.

           +-----------+            +-----------+
           |           |            |           |
--(UART)-->|  task1    |--(Queue)-->|   task2   |
           |           |            |           |
           +-----------+            +-----------+