espressif/esp-idf #15894: If the read buffer of i2c_master_transmit_receive() is initialized, it wasn't the register value but the init value.

WaitNothing, 2025-05-02:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

```cpp
 //i2c.hpp
#ifndef _I2C_HPP_
#define _I2C_HPP_

#include "driver/i2c_master.h"

class I2C{
private:
    i2c_master_bus_handle_t bus_handle;
    i2c_master_dev_handle_t dev_handle;

public:
    I2C(i2c_port_num_t port_num, gpio_num_t sda_io, gpio_num_t scl_io, uint8_t device_address);
    ~I2C();
    bool write(uint8_t* buff_w, size_t len);
    bool read(uint8_t reg_addr, uint8_t* buff_r, size_t len_r);
};

#endif //_I2C_HPP_
```

```cpp
//i2c.cpp
#include "i2c.hpp"

I2C::I2C(i2c_port_num_t port_num, gpio_num_t sda_io, gpio_num_t scl_io, uint8_t device_addr){
    i2c_master_bus_config_t i2c_mst_cfg = {
        .i2c_port = I2C_NUM_0,
        .sda_io_num = sda_io,
        .scl_io_num = scl_io,
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .glitch_ignore_cnt = 7,
        .intr_priority = 0,           // 必须设为0！
        .trans_queue_depth = 8,
        .flags = {
            .enable_internal_pullup = true // 关键修正项！启用内部上拉
        }
    };

    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = device_addr,
        .scl_speed_hz = 10000,
    };

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_cfg, &bus_handle));
    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
}

I2C::~I2C(){
    i2c_master_bus_rm_device(dev_handle);
    i2c_del_master_bus(bus_handle);
}

bool I2C::write(uint8_t* buff_w, size_t len){
    int ret = i2c_master_transmit(dev_handle, buff_w, len, -1);
    if(ret != ESP_OK){
        return false;
    }
    return true;
}

bool I2C::read(uint8_t reg_addr, uint8_t* buff_r, size_t len_r){
    esp_err_t ret = i2c_master_transmit_receive(dev_handle, &reg_addr, 1, buff_r, 2, -1);
    printf("buff_r[0]: %02X, buff_r[1]: %02X, ret = %d\n", buff_r[0], buff_r[1], ret);
    if(ret != ESP_OK){
        return false;
    }
    return true;
}
```
```cpp
//as5600.hpp
#ifndef _AS5600_HPP_
#define _AS5600_HPP_

#include "i2c.hpp"

#define AS5600_RAW_ANGLE_REG_HIGH 0x0C
#define AS5600_RAW_ANGLE_REG_LOW 0x0D
class As5600{
private:
    I2C *i2c;
public:
    As5600(i2c_port_num_t port_num, gpio_num_t sda_io, gpio_num_t scl_io, uint8_t device_addr);
    ~As5600();

    float getAngle();
};

#endif //_AS5600_HPP_
```
```cpp
//as5600.cpp
#include "as5600.hpp"

As5600::As5600(i2c_port_num_t port_num, gpio_num_t sda_io, gpio_num_t scl_io, uint8_t device_addr){
    i2c = new I2C(port_num, sda_io, scl_io, device_addr);
}

As5600::~As5600(){
    delete(i2c);
}

float As5600::getAngle(){
    uint8_t buff_r[2]={0xff,0xff};
    uint16_t raw_angle = 0;
    i2c -> read(0x0C, buff_r, 2);
    // printf("buff_r[0]: %02X, buff_r[1]: %02X\n", buff_r[0], buff_r[1]);
    raw_angle = (buff_r[0] << 8) | (buff_r[1]);

    return ((float)raw_angle * 360.0) / 4096.0;
}
```
```cpp
#include "driver/i2c_master.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "as5600.hpp"
#include "esp_log.h"

static void as5600_task(void* pvParameters){
    As5600 leftEncoder(0, GPIO_NUM_19, GPIO_NUM_18, 0x36);

    while(true){
        float angle = leftEncoder.getAngle();
        ESP_LOGI("AS5600", "Angle: %.2f", angle);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}

extern "C" void app_main(){
    //xTaskCreatePinnedToCore(&as5600_task, "As5600_task", 4096, NULL, 5, NULL, 0);
    as5600_task(NULL);
}
```
If the read buffer of i2c_master_transmit_receive() is initialized, it wasn't the register value but the init value.
```cpp
float As5600::getAngle(){
    uint8_t buff_r[2]={0xff,0xff};
    uint16_t raw_angle = 0;
    i2c -> read(0x0C, buff_r, 2);
    // printf("buff_r[0]: %02X, buff_r[1]: %02X\n", buff_r[0], buff_r[1]);
    raw_angle = (buff_r[0] << 8) | (buff_r[1]);

    return ((float)raw_angle * 360.0) / 4096.0;
}
```
The `c` version is correct wether initialized or not:
```c
#include "driver/i2c_master.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define AS5600_ADDR 0x36
#define AS5600_RAW_ANGLE_REG_HIGH 0x0C
#define AS5600_RAW_ANGLE_REG_LOW 0x0D

i2c_master_bus_handle_t bus_handle;
i2c_master_dev_handle_t dev_handle;

void as5600_init(i2c_port_num_t port_num, gpio_num_t sda_io, gpio_num_t scl_io){
    i2c_master_bus_config_t i2c_mst_cfg = {
        .clk_source = I2C_CLK_SRC_DEFAULT,
        .i2c_port = port_num,
        .scl_io_num = scl_io,
        .sda_io_num = sda_io,
        .glitch_ignore_cnt = 7,
        .flags.enable_internal_pullup = true,
    };

    ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_cfg, &bus_handle));

    i2c_device_config_t dev_cfg = {
        .dev_addr_length = I2C_ADDR_BIT_LEN_7,
        .device_address = AS5600_ADDR,
        .scl_speed_hz = 100000,
    };

    ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
}

uint16_t as5600_read_angle(){
    uint8_t buf_W[2] = {AS5600_RAW_ANGLE_REG_HIGH};
    uint8_t buf_R[2] = {0, 0};
    uint16_t raw_angle;

    ESP_ERROR_CHECK(i2c_master_transmit_receive(dev_handle, buf_W, 1, buf_R, 2, -1));
    printf("buf_R[0]: %02X buf_R[1]: %02X\n", buf_R[0], buf_R[1]);
    raw_angle = (buf_R[0] << 8 )| (buf_R[1]);
    return raw_angle;
}

float as5600_convert_to_angle(uint16_t raw_angle) {
    return ((float)raw_angle * 360.0) / 4096.0;
}

void app_main(){
    i2c_port_num_t port_num = I2C_NUM_0;
    as5600_init(port_num, 19, 18);// Initialize I2C with SDA on GPIO 19 and SCL on GPIO 18

    while (1) {
        uint16_t raw_angle = as5600_read_angle();
        float angle = as5600_convert_to_angle(raw_angle);
        printf("Angle: %.2f degrees\n", angle);
        vTaskDelay(500 / portTICK_PERIOD_MS);
    }
}
```

### What is the actual behavior?

output:
```
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (1350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (2350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (3350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (4350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (5350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (6350) AS5600: Angle: 5759.91
buff_r[0]: FF, buff_r[1]: FF, ret = 0
I (7350) AS5600: Angle: 5759.91
```

### Steps to reproduce.

1. initialize the read buffer
2. not initialize the read buffer

### Debug Logs.

```plain

```

mythbuster5, 2025-05-08:
@WaitNothing Please remove `.trans_queue_depth = 8,` in your cpp wrapper/

mythbuster5, 2025-05-12:
Because if with `trans_queue_depth`, function returns doesn't mean the read operation finish. So that's why you got a wrong value. If no other update, I will close this issue.

closed 2025-05-12 by mythbuster5