espressif/esp-idf #15895: Support for access to LP module's I2C module from the Xtensa CPU (ESP32-S3).

eriksl, 2025-05-02:
### Is your feature request related to a problem?

The ESP32-S3 has a LPM (one or two, I am sure if they're actually separate entities, the state machine and the RISC-V processor) that also features an I2C module. I'd like to use this module in "normal running" mode as well, from the Xtensa CPU. As far as I can see, this is possible by reading/writing the relevant registers from the Xtensa CPU, but also there doesn't seem to be support within ESP-IDF to do so.

So that's my feature request.

sudeep-mohanty, 2025-05-02:
Hi @eriksl,
You can use the LP module's I2C peripheral (known as the RTC I2C controller), from the main cpu (a.k.a, the Xtensa core) using the APIs defined [here](https://github.com/espressif/esp-idf/blob/master/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h). Although the peripheral is designed to be used from the ULP-RISCV core, [this example](https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/examples/system/ulp/ulp_riscv/i2c/main/ulp_riscv_rtc_i2c_example_main.c#L76) demonstrates how the peripheral is used from the main core as well, before going in to sleep and handing over the control of the RTC I2C peripheral to the ULP-RISCV core.

eriksl, 2025-05-02:
I found this file, but I assumed this code could only be called from code to be compiled for the RISC-V ulp core?

I will try it!

eriksl, 2025-05-03:
At least I got it to compile! Which was kind of a challenge, because IDF refused to look for the relevant include files. Apparently you need to add "soc ulp" to the list of COMPONENTS in the top level CMakeLists.txt file (the way it's done in the example doesn't work (anymore)). After that, you'll see the relevant options for ULP in the menuconfig and after that, you can use the include files. If you do it like in the example, the "normal" components, like "main", "bt" etc. are no longer available, even if you add them explicitly.

So, next step: connect it up and try it. I'll use a protocol analyser to make sure things go like I want.

Can I say it's daft we're required to first set at least one byte, the first, as a "reg addr" before being able to send any data? As far as I can see from the hardware guide, this is not required by the hardware. So **[feature request]** please add functions for:
- sending "just" data
- receiving "just" data (without sending any)
- if at all possible, implement a cycle like <start><send><repeated start><receive><stop> which is used very frequently for I2C.
- probing a device (send the address only, no actual reading or writing)

Of course the ultimate solution would be to fully integrate this into the "main" I2C implementation, just adding a third I2C bus. But I guess there would be too few people using it to warrant the effort!

eriksl, 2025-05-03:
@sudeep-mohanty thanks for your help so far!

eriksl, 2025-05-04:
It seems to work, no more help required I think 👍

closed 2025-05-04 by eriksl