espressif/esp-idf #15899: Can a form of esp_log_set_vprintf or similar function be created that passed the tag or log level?

mbratch, 2025-05-02:
### Is your feature request related to a problem?

I would like to intercept the logging using `esp_log_set_vprintf` and send, for example, to a syslog server. The syslog service understands the concept of "level" of message (e.g., error, warning, info, debug, etc). I would like to set the appropriate level in the syslog message. However, this information is not passed to `esp_log_set_vprintf`.

### Describe the solution you'd like.

Is there a function that already does this, or perhaps such a function could be created? If it passed the log message tag that would be sufficient because then I could call `esp_log_level_get`.

### Describe alternatives you've considered.

I've briefly considered parsing the text of the log message, knowing it is prefixed with a tag, but this is of course not a reliable way of obtaining the tag (i.e., not a good idea).

mbratch, 2025-05-05:
@sudeep-mohanty thank you! Either log level or the tag would suffice since log level can be obtained from the tag.

igrr, 2025-05-05:
It would be really nice if we allowed calling custom log formatting/storage functions over here:

https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/components/log/src/log.c#L58-L73

and not only the bundled `esp_log_format` and `esp_log_format_binary`. That would satisfy the feature request and also allow for development of out-of-tree log handling components (e.g. "log to Flash", "log to esp-insights", etc.)

hachi, 2025-08-12:
Hello, looking for traction on this issue.

In #10494 it was also stated that newline is supplied by caller, but formatting is to be handled by vprintf-like function.

Documentation doesn't match reality, as newline is part of formatting.

As implemented today to create syslog-like component for esp logging I have hook into vprintf and then:
- Strip out colors if included
- Read first character to find log level and coerce to syslog-priority field
- Read next part to strip out time and convert to syslog-time format
- Read next part to get TAG and coerce to app name in syslog format
- Remove any newlines, optionally split message on newlines and output multiple messages

The existing esp_log system doesn't allow a method to easily add fields to the middle of a log line like:
- Task/Thread name
- Calling file/line/function
- etc

The only option you have by configuring vprintf is to prepend/append or fully parse the line to actually change the format.

We might actually be able to do most of these things if a hook was supplied for `esp_log` or `esp_log_va`.

`esp_log_set_vprintf` really only lets us send the output to a diferent filehandle/uart