espressif/esp-idf #15900: ESP32-S3 Bluedroid Scan extended for coded phy

Olfox59, 2025-05-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi

I'm using esp-idf v5.4.1 with my esp32-s3.
My firmware is a ble client and server, that use bluedroid stack.

I worked with ble 5.0 1M phy before but for distance and range reasons, i have to move to coded Phy ( long range ble).

I upated my code to use extended scan et coded phy parametre.

1) First, i'm not sur to be in coded phy because of the return value by the function:

` esp_ble_gap_set_preferred_default_phy(ESP_BLE_GAP_PRI_PHY_CODED, ESP_BLE_GAP_PRI_PHY_CODED);`

> I (2715) Gap CallBack: PHY default prefered complete: status: 0, tx_phy=206, rx_phy=63

2) In client mode, i need to connect to nrf52833 advertising on coded phy pri and sec properly ( checked with nrfConnect mobil app).

So i'm using now 

```
    //Configuration  codedphy scanner
    esp_ble_ext_scan_params_t ext_scan_params = {
        .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
        .filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_duplicate = BLE_SCAN_DUPLICATE_ENABLE,    //juste pour le debug ne pas afficher trop de fois la meme adresse mac
        .uncoded_cfg.scan_type = BLE_SCAN_TYPE_ACTIVE,
        .coded_cfg.scan_type = BLE_SCAN_TYPE_ACTIVE,
        .coded_cfg.scan_interval = 0x50,
        .coded_cfg.scan_window = 0x30,

    };
    esp_ble_gap_set_ext_scan_params(&ext_scan_params);
```

But i get the error from bt stack:

> E (2735) BT_BTM: LE ES SetParams: cmd err=0xc
> E (2735) BT_HCI: CC evt: op=0x2041, status=0xc

Instead of `esp_ble_gap_start_scanning(duration);` i'm using now :  `esp_ble_gap_start_ext_scan(0xffff, 1)
`

I would like to be in phy 1M for my server mode (conenxion with smartphone) , and coded phy for client mode.

Here is my full log for information:

> I (2525) main_task: Returned from app_main()
> I (2525) Task_BLE: ** In task BLE **
> I (2525) Task_BLE: stack heap: 2087396 bytes
> I (2525) BLE_INIT: BT controller compile version [d74042a]
> I (2535) BLE_INIT: Bluetooth MAC: 30:30:f9:50:be:9e
> I (2535) init_ble_gattc_gatts: ** esp_bt_controller_enable **
> I (2545) phy_init: phy_version 700,8582a7fd,Feb 10 2025,20:13:11
> I (2585) init_ble_gattc_gatts: ** esp_bluedroid_init **
> I (2605) init_ble_gattc_gatts: ** esp_bluedroid_enable **
> I (2625) init_ble_gattc_gatts: ** esp_ble_gap_register_callback **
> I (2635) Server_event: REGISTER_APP_EVT, status 0, app_id 0
> 
> I (2635) Server_event: advertising name: ESP32S3-ble
> 
> W (2635) BT_BTM: BTM_BleWriteAdvData, Partial data write into ADV
> I (2645) Gap CallBack: ESP_GAP_BLE_SCAN_RSP_DATA_SET_COMPLETE_EVT
> Start Advertising...
> I (2645) Server_event: CREATE_SERVICE_EVT, status 0,  service_handle 40
> I (2665) Gap CallBack: Advertising start successfully
> I (2665) Server_event: SERVICE_START_EVT, status 0, service_handle 40
> I (2665) Server_event: ADD_CHAR_EVT, status 0,  attr_handle 42, service_handle 40
> I (2675) Server_event: ADD_DESCR_EVT, status 0, attr_handle 43, service_handle 40
> I (2685) Gatt cb: ESP_GATTC_REG_EVT
> I (2685) Client_A_event_handler: REG_EVT
> I (2685) Gatt cb: ESP_GATTC_REG_EVT
> I (2695) Client_B_event_handler: REG_EVT
> I (2695) Gap CallBack:  ! ESP_GAP_BLE_SCAN_PARAM_SET_COMPLETE_EVT !
> I (2705) etat:: SCAN
> try to start scan ext!
> I (2705) Gap CallBack:  ! ESP_GAP_BLE_SCAN_START_COMPLETE_EVT ! 
> I (2715) Gap CallBack: Scan start success
> I (2715) Gap CallBack: PHY default prefered complete: status: 0, tx_phy=206, rx_phy=63
> E (2725) BT_BTM: LE ES SetParams: cmd err=0xc
> E (2725) BT_HCI: CC evt: op=0x2041, status=0xc
> I (2735) etat:: INIT
> I (2735) etat:: INIT
> my_rf_power adv: 14
> my_rf_power scan: 14
> I (2735) Gap CallBack: ESP_GAP_BLE_SCAN_RESULT_EVT
> I (2745) Gap CallBack: 39 7a ce 87 c6 6f
> my_rf_power default: 14
> my_rf_power con hdl0: 255
> new my_rf_power: 14
> my_rf_power adv: 14
> my_rf_power scan: 14
> my_rf_power default: 14
> my_rf_power con hdl0: 255
> MEM RESTANTE : 5856 mots
> I (3615) Gap CallBack: ESP_GAP_BLE_SCAN_RESULT_EVT
> I (3615) Gap CallBack: 71 9c 8e 26 50 2d

Thank you for your support :)

Olfox59, 2025-05-04:
This code:

```
    esp_ble_ext_scan_params_t ext_scan_params = {
        .own_addr_type = BLE_ADDR_TYPE_PUBLIC,
        .filter_policy = BLE_SCAN_FILTER_ALLOW_ALL,
        .scan_duplicate = BLE_SCAN_DUPLICATE_DISABLE,    //juste pour le debug ne pas afficher trop de fois la meme adresse mac
        //.coded_cfg = ESP_BLE_GAP_EXT_SCAN_CFG_CODE_MASK,
        //.scan_type = ESP_BLE_SCAN_TYPE_ACTIVE,
        //.uncoded_cfg.scan_type = BLE_SCAN_TYPE_ACTIVE,
        .coded_cfg.scan_type = BLE_SCAN_TYPE_ACTIVE,
        .coded_cfg.scan_interval = 0x50,
        .coded_cfg.scan_window = 0x30,

    };
    esp_ble_gap_set_ext_scan_params(&ext_scan_params);
```

Generate this error:

> 
> E (2715) BT_BTM: LE ES SetParams: cmd err=0xc
> E (2715) BT_HCI: CC evt: op=0x2041, status=0xc

But the error is always the same independtly from the content of ext_scan_param. At the moement i call the function, the error pops up.

It seems that i have to move to nimble that support in a better way the coded phy , is it true ?

Olfox59, 2025-05-05:
I finally decided to switch to nimBle.

But i meet another issue, mutli client seems to not be possible with esp32 s3 according to the documentation.

I had no issue with Bluedroid and 2 client + ble server enable at the same time in 1Mbps.

Change for another chip like h2 or c6 series is not really possible because of custom hardware and I need power for lvgl rendering (multicore help with that).

DO you have any tips r recommandations to suround this issue please ?

Thank you !

esp-zhp, 2025-05-19:
Did switching to NimBLE resolve the issue?

esp-zhp, 2025-05-19:
If you're using the Bluedroid host, you can refer to the example below.

https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble_50/ble50_security_client