espressif/esp-idf #15901: ULP I2C read and write functions that do not send the "sub reg" byte.

eriksl, 2025-05-04:
### Is your feature request related to a problem?

The problem is that some I2C sensors (notably **htu21** and **hdc1080**) require a read operation without any write operation to obtain the current data. When any write is issued, the measuring cycle is restarted, so no data is available when switching to read, resulting in a NAK from the device.

### Describe the solution you'd like.

Make sending of the "sub reg" byte optional. Or remove the "sub reg" thing altogether, because it can be replaced writing the byte explicitly from the sending data buffer. In that case an extra function should be added, where data can be written and read in one transaction using a repeated start condition (like the functions for the "normal"  (main core) I2C module do as well).

### Describe alternatives you've considered.

For this issue, afaik, there is no alternative.

BitsForPeople, 2025-05-04:
Don't know which ESP/ULP you're referring to, but on the ESP32-S3 (and the original ESP32) this seems to be implemented in hardware, which makes the ["RTC I2C peripheral incompatible with certain I2C devices or sensors which do not need any sub-register to be programmed."](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32s3/api-reference/system/ulp-risc-v.html#rtc-i2c).

eriksl, 2025-05-05:
I am familiar with this piece of documentation. The text doesn't state, though, whether it's a software or a hardware limitation. If it is the latter, I'd request the text to be change to reflect this.

And of course for future designs to have a more generic ULP I2C controller, of course. I am not sure why the "obligatory" "register sub address" should be a handled specially, like this. A generic write-repeatedACK-read cycle, where all lengths are optional (i.e. writing or reading is optional, and if applicable, the repeatedACK), would be perfect.

sudeep-mohanty, 2025-05-05:
@eriksl Unfortunately, the programming of the I2C slave sub-register address is mandatory and is a limitation of the RTC I2C peripheral (or in other words, the hardware). The driver simply respects this limitation which, unfortunately, could break compatibility with certain I2C devices or sensors.

This limitation has been addressed in more recent iterations of the ULP processors. For instance, the LP-I2C (low-power I2C) peripheral on the ESP32-C6 (and future targets) do not have this restriction.

eriksl, 2025-05-05:
Good to hear the issue has been addressed.

BTW I tried modifying the code so (according to the commands sent) there would be no "sub register address" sent. But it doesn't work as expected. So confirmed, it's a hardware issue.

ESP-Marius, 2025-05-07:
Closing this one as there isnt really anything we do to address this in ESP-IDF.

closed 2025-05-07 by ESP-Marius

eriksl, 2025-05-07:
I am not 100% happy with the current documentation, it confused me. Please accept my clarification: https://github.com/espressif/esp-idf/pull/15917