espressif/esp-idf #15902: Unable to install console component

mbratch, 2025-05-04:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I was expecting to install the console component with the command line:

`idf.py add-dependency console`.

### What is the actual behavior?

When I run the command `idf.py add-dependency console` I get the following error:
```
$ idf.py add-dependency console
Executing action: add-dependency
ERROR: Component "espressif/console" not found
```
The console component is described in the [documentation](https://docs.espressif.com/projects/esp-idf/en/v5.4.1/esp32/api-reference/system/console.html), but I could not find any notes regarding how it is installed.

### Steps to reproduce.

1. Enter the command `idf.py add-dependency console`
2. Observe error: "Component "espressif/console" not found"

### Build or installation Logs.

```plain

```

sudeep-mohanty, 2025-05-05:
Hi, @mbratch Please add the dependency directly in your `main` component `CMakeLists.txt` file as shown [here](https://github.com/espressif/esp-idf/blob/38628f98b9000ded87827dccbd8a5a1827392cfb/examples/system/console/advanced/main/CMakeLists.txt#L2).

mbratch, 2025-05-05:
@sudeep-mohanty thanks for the suggestion.
Taking this approach throws my project into a bit of mayhem.

For example, now I get:

```
fatal error: driver/gpio.h: No such file or directory
    3 | #include <driver/gpio.h>
      |          ^~~~~~~~~~~~~~~
```
So now perhaps several other things need to be explicitly specified that didn't need to be before?

mbratch, 2025-05-05:
@sudeep-mohanty thanks for the suggestion.
Taking this approach throws my project into a bit of mayhem.

I'm not getting "undefined reference to `app_main`".

mbratch, 2025-05-05:
@sudeep-mohanty thanks very much for the suggestion.

I'm now getting "undefined reference to app_main" at the linking stage.

mbratch, 2025-05-06:
@sudeep-mohanty thanks, that worked, but I had to hunt down and add in all the other components I was using once I used the PRIV_REQUIRES. I had to add all of them to the list. I'm not sure how the tools knew I needed all the other components to this point before I decided to use console. The project always built.

Is there a reason for the lack of consistency between how some components need to be included?

ESP-Marius, 2025-05-06:
The `main` component is a special case, as by default it REQUIRES all components by default, see [docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#main-component-requirements).

By manually adding a `REQUIRE` this behavior is overwritten, so you'll need to also add all the other components you are using.

Console is a core component that comes with ESP-IDF, so it doesnt need to use the component manager to download and add it. 

But we could consider adding a helpful error message if you try to use the component manager to add a component that already exists in IDF locally 👍

closed 2025-05-06 by mbratch