espressif/esp-idf #15904: ULP I2C peripheral on ESP-S2 and ESP-S3 does not report communication errors back to the application [patch included]

eriksl, 2025-05-05:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Not relevant here

### Power Supply used.

USB

### What is the expected behavior?

Expected: when I read or writes to the ULP I2C bus and the I2C slave device isn't responding (i.e. sends no ACK) I expect to able to be notified from this fact, in my application, so I can handle the situation.

### What is the actual behavior?

Actual: the relevant functions are made of "void" return type:

```
void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size);
void ulp_riscv_i2c_master_write_to_device(uint8_t *data_wr, size_t size);
```

so the application can never know of the status. Also in case of an issue, all sorts of debugging information is dumped into the log instead. This makes it impossible to **probe** for devices.

The code has two other issue I'd like to have fixed:

- it silently ignores the issue when zero bytes are requested to be read or written. The I2C cannot handle this, but no error is reported.
- the "write" function lacks a "const" qualifier to the byte array argument

The main issue (void return value) can be fixed without impact for existing applications. The other two issues have minimal impact. So I request this fix to be included into the next maintenance release (v5.4.2).

Here is the patch:

```
diff --git a/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h b/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h
index 503fa7b20c..70847a4ab4 100644
--- a/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h
+++ b/components/ulp/ulp_riscv/include/ulp_riscv_i2c.h
@@ -115,7 +115,7 @@ void ulp_riscv_i2c_master_set_slave_reg_addr(uint8_t slave_reg_addr);
  * @param data_rd       Buffer to hold data to be read
  * @param size          Size of data to be read in bytes
  */
-void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size);
+esp_err_t ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size);
 
 /**
  * @brief Write to I2C slave device
@@ -125,7 +125,7 @@ void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size);
  * @param data_wr       Buffer which holds the data to be written
  * @param size          Size of data to be written in bytes
  */
-void ulp_riscv_i2c_master_write_to_device(uint8_t *data_wr, size_t size);
+esp_err_t ulp_riscv_i2c_master_write_to_device(const uint8_t *data_wr, size_t size);
 
 /**
  * @brief Initialize and configure the RTC I2C for use by ULP RISC-V
diff --git a/components/ulp/ulp_riscv/ulp_riscv_i2c.c b/components/ulp/ulp_riscv/ulp_riscv_i2c.c
index 138d479642..26212ec111 100644
--- a/components/ulp/ulp_riscv/ulp_riscv_i2c.c
+++ b/components/ulp/ulp_riscv/ulp_riscv_i2c.c
@@ -316,15 +316,15 @@ void ulp_riscv_i2c_master_set_slave_reg_addr(uint8_t slave_reg_addr)
  * | Slave  |        |         |  ACK   |        |   ACK  |        |         |   ACK  |  DATA  |        |  DATA  |        |        |
  * |--------|--------|---------|--------|--------|--------|--------|---------|--------|--------|--------|--------|--------|--------|
  */
-void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size)
+esp_err_t ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size)
 {
     uint32_t i = 0;
     uint32_t cmd_idx = 0;
     esp_err_t ret = ESP_OK;
+	esp_err_t result = ESP_OK;
 
     if (size == 0) {
-        // Quietly return
-        return;
+        return(ESP_ERR_INVALID_ARG);
     }
 
     /* By default, RTC I2C controller is hard wired to use CMD2 register onwards for read operations */
@@ -379,11 +379,9 @@ void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size)
             /* Clear the Rx data interrupt bit */
             SET_PERI_REG_MASK(RTC_I2C_INT_CLR_REG, RTC_I2C_RX_DATA_INT_CLR);
         } else {
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: Read Failed!");
             uint32_t status = READ_PERI_REG(RTC_I2C_INT_RAW_REG);
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x%"PRIx32"", status);
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: RTC I2C Status Reg 0x%"PRIx32"", READ_PERI_REG(RTC_I2C_STATUS_REG));
             SET_PERI_REG_MASK(RTC_I2C_INT_CLR_REG, status);
+			result = ESP_ERR_INVALID_RESPONSE;
             break;
         }
     }
@@ -393,6 +391,8 @@ void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size)
     /* Clear the RTC I2C transmission bits */
     CLEAR_PERI_REG_MASK(SENS_SAR_I2C_CTRL_REG, SENS_SAR_I2C_START_FORCE);
     CLEAR_PERI_REG_MASK(SENS_SAR_I2C_CTRL_REG, SENS_SAR_I2C_START);
+
+	return(result);
 }
 
 /*
@@ -412,15 +412,15 @@ void ulp_riscv_i2c_master_read_from_device(uint8_t *data_rd, size_t size)
  * | Slave  |        |         |  ACK   |        |   ACK  |        |   ACK  |        |   ACK  |        |
  * |--------|--------|---------|--------|--------|--------|--------|--------|--------|--------|--------|
  */
-void ulp_riscv_i2c_master_write_to_device(uint8_t *data_wr, size_t size)
+esp_err_t ulp_riscv_i2c_master_write_to_device(const uint8_t *data_wr, size_t size)
 {
     uint32_t i = 0;
     uint32_t cmd_idx = 0;
     esp_err_t ret = ESP_OK;
+	esp_err_t result = ESP_OK;
 
     if (size == 0) {
-        // Quietly return
-        return;
+        return(ESP_ERR_INVALID_ARG);
     }
 
     /* By default, RTC I2C controller is hard wired to use CMD0 and CMD1 registers for write operations */
@@ -455,11 +455,9 @@ void ulp_riscv_i2c_master_write_to_device(uint8_t *data_wr, size_t size)
             /* Clear the Tx data interrupt bit */
             SET_PERI_REG_MASK(RTC_I2C_INT_CLR_REG, RTC_I2C_TX_DATA_INT_CLR);
         } else {
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: Write Failed!");
             uint32_t status = READ_PERI_REG(RTC_I2C_INT_RAW_REG);
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: RTC I2C Interrupt Raw Reg 0x%"PRIx32"", status);
-            ESP_EARLY_LOGE(RTCI2C_TAG, "ulp_riscv_i2c: RTC I2C Status Reg 0x%"PRIx32"", READ_PERI_REG(RTC_I2C_STATUS_REG));
             SET_PERI_REG_MASK(RTC_I2C_INT_CLR_REG, status);
+			result = ESP_ERR_INVALID_RESPONSE;
             break;
         }
     }
@@ -469,6 +467,8 @@ void ulp_riscv_i2c_master_write_to_device(uint8_t *data_wr, size_t size)
     /* Clear the RTC I2C transmission bits */
     CLEAR_PERI_REG_MASK(SENS_SAR_I2C_CTRL_REG, SENS_SAR_I2C_START_FORCE);
     CLEAR_PERI_REG_MASK(SENS_SAR_I2C_CTRL_REG, SENS_SAR_I2C_START);
+
+	return(result);
 }
 
 esp_err_t ulp_riscv_i2c_master_init(const ulp_riscv_i2c_cfg_t *cfg)
```
Thanks!

### Steps to reproduce.

Not relevant.

### Debug Logs.

```plain

```

sudeep-mohanty, 2025-05-05:
Hi @eriksl,
Thank you for the suggestions and the contribution! 

Changing the signature of the `ulp_riscv_i2c_master_write_to_device/read_from_device()` to return `esp_err_t` instead of `void` would be considered a breaking change which we cannot do in a minor version release of ESP-IDF. Hence, this is something that would probably have to wait until ESP-IDF release v6.0.

Having said that, it may still possible to propagate transaction errors back to the application by having a global error status and providing a getter function. Although it is not the most intuitive solution, it would help solve your problem as well as not induce a breaking change on the API.

eriksl, 2025-05-05:
Not a single line of existing code needs to be modified, it's perfectly fine for any code to ignore the esp_err_t type return value, just like it's mandatory for void typed functions. So I do not agree there.

eriksl, 2025-05-07:
Nice! Very much appreciated!

I was going to suggest to add an alternative interface (with patch) as additional functions for this:

- ulp_riscv_i2c_master_write_to_device_with_status
- ulp_riscv_i2c_master_read_from_device_with_status

and the existing functions would become wrappers to these functions.

But now I don't need to 👍

eriksl, 2025-09-14:
Hmmm just a little disappointed this apparently went into development or v6, while it could have been in v5.5.