espressif/esp-idf #15905: Example showing ADC and PWM synchronization using Event Task Matrix

5ami, 2025-05-05:
### Is your feature request related to a problem?

This is not exactly a feature. But a suggestion for an example that can be added. 
Synchronization between ADC and PWM ( Trig an ADC sample at specific PWM counter value) Is an important feature to have in power electronics applications. ETM peripheral seems to be able to handle this seamlessly by hardware.  I hope to see an example showcasing this.

suda-morris, 2025-05-06:
The ETM driver development of ADC is already in our backlog, thank you for your feedback, and we will provide the corresponding example when the driver design is completed.

ildar-belkin, 2026-10-25:
Hello! I noticed that the changes haven't been applied yet to either 5.5.6 or 6.1. Are there any updates on when this might be planned? Thanks!