espressif/esp-idf #15907: ESP32 emmc/sdmmc returned 0x107 when using ledc with fade

holznan, 2025-05-06:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Espressif SoC revision.

ESP32-WROVER-E

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 5V

### What is the expected behavior?

Hello espressif team,

we have an application which should log errors of a system to an emmc module which is equipped over sdmmc.
Therefore i use fatfs. When some kind of error appears, we open a file on the emmc write the error log into it and close it. 
Additional the application can control two PWM solenoids. These are connected to an Infineon BTS which is controlled over ledc with fade. 

### What is the actual behavior?

As soon as i init/use the ledc with fade i get errors on the sdmmc peripheral. 
Every sdmmc command fails and returns 0x107
for example: sdmmc_host_start_command returned 0x107
                     sdmmc_read_sector_dma: sdmmc_send_cmd returned 0xFFFFFFFF, failed to get status (0x107)
So no access to the emmc module is possible any more.

### Steps to reproduce.

I made a small sample project to reproduce the error (see code below). It is in general the ledc with fade example from espressif and i added a part from the fatfs example. 
When i comment the line out for "app_init_ledc()" in the app_main function. emmc just working as expected. 
When using the init for the ledc with fade the errors occure immediately.

Here is the code:

/* LEDC (LED Controller) fade example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/semphr.h"
[... 512 lines cut ...]
I (269) heap_init: At 3FFB2E00 len 0002D200 (180 KiB): DRAM
I (274) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (279) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (285) heap_init: At 4008CE04 len 000131FC (76 KiB): IRAM
I (292) spi_flash: detected chip: gd
I (293) spi_flash: flash io: dio
W (296) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (309) main_task: Started on CPU0
I (319) main_task: Calling app_main()
I (319) example: Initializing eMMC
I (319) example: Using SDMMC peripheral
I (319) example: Mounting filesystem
I (359) example: Filesystem mounted
I (359) example: Create a new file /eMMC/Testfile.csv
Name: AJNB4R
Type: MMC
Speed: 40.00 MHz (limit: 52.00 MHz)
Size: 14910MB
CSD: ver=3, sector_size=512, capacity=30535680 read_bl_len=9
EXT CSD: bus_width=1
I (369) example: Opening existing ErrorLog file /eMMC/ErrorLog.csv
I (379) example: read last block of file:
I (379) example: Line Ending not found
I (389) example: Close ErrorLog File
```

diplfranzhoepfinger, 2025-05-06:
well done @holznan

diplfranzhoepfinger, 2025-05-06:
is it related with this one: 
https://github.com/espressif/esp-idf/issues/5147

wanckl, 2025-05-20:
@holznan Hi, Guess it is hardware issue, the PWM noise breaks the eMMC communication.

- Can you try init LEDC but start it after some delay, to see if eMMC failed at the moment which LEDC start not LEDC init.
- Normally start LEDC but don't connect to your BTS, even disconnect power supply of BTS side to avoid high power noise.

holznan, 2025-05-22:
@wanckl Hi, thanks for the answer. 

I tried some things with the sample project above. 

When i only call "ledc_timer_config(&ledc_timer)" in the LEDC init, everything works fine with eMMC. 
But as soon as i call "ledc_channel_config(&ledc_channel[ch])" the eMMC stops working. 

So the problem seems to be already in the LEDC init. 
I don´t even start to use LEDC.

songruo, 2025-06-03:
@holznan Sorry for the late reply. The call to `ledc_channel_config` already starts the LEDC output. Are you sure that your LEDC IO is not conflict with the SDMMC IOs? If you can upgrade IDF version to v5.5 or master branch, it might become more clear, the IO conflict is checked internally. You would see a warning log if IO is used conflicting with each other.

Alvin1Zhang, 2025-06-30:
@holznan Thanks for reporting, would you please help share if any updates for the issue? Thanks.

holznan, 2025-07-02:
Hello, 
i will try as soon as possible but at the moment i am busy. I hope i can test this by the end of next week.

diplfranzhoepfinger, 2025-07-16:
@wanckl 
@Alvin1Zhang 
@songruo 

i have @holznan on the Phone. 
he compiled same Project in ESP-IDF Release v5.4.0 in Windows
and ESP-IDF Release v5.4.1 in Linux. 

in Linux Error is gone. 

further Analysis is taken. Hold on!

holznan, 2025-07-17:
@wanckl
@Alvin1Zhang
@songruo 

Hi everyone, 

after testing i can confirm that the issue is gone with esp-idf v5.4.1 and higher. 
Thanks a lot for your assistance. 

Best regards!